stack-of-tasks/pinocchio #1493: Failing to compile from source (ubuntu 14.04 ,  g++ 9.4.0)

vincentberenz, 2021-08-12:
I am trying to compile pinocchio from source. Alas, I am for the moment stuck with  ubuntu 14.04 (because of some xenomai kernel that we fail for the moment to compile on newer versions of ubuntu).

I updated g++ to 9.4 and  cmake to 3.15.3 . I also installed eigenpy after compilation from source. boost is at version 1.54.0

```cmake``` runs with success:

```
-- Package version (ROS package.xml): 2.6.3
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") found components:  doxygen dot 
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Since urdfdom >= 1.0.0, the default C++ standard is C++11. The project is then compiled with C++11 standard.
-- Found Boost: /usr/include (found version "1.54.0") found components:  filesystem serialization system 
-- The Python bindings of Pinocchio will be compiled along the main library. If you want to disable this feature, please set the option BUILD_PYTHON_INTERFACE to OFF.
-- Found Python2: /usr/local/bin/python (found version "2.7.18") found components:  Interpreter Development 
-- PythonLibraryDirs: /usr/local/lib
-- PythonLibVersionString: 2.7.18
-- Python site lib: lib/python2.7/site-packages
-- Found Boost: /usr/include (found version "1.54.0") found components:  python27 
-- Boost_PYTHON_LIBRARY: /usr/lib/x86_64-linux-gnu/libboost_python-py27.so
-- eigenpy FOUND. eigenpy at /usr/local/lib/libeigenpy.so
-- Python compiler: CPython
-- Found Boost: /usr/include (found version "1.54.0") found components:  unit_test_framework 
-- Doxygen rendering: using MathJax backend
-- Configuring done
-- Generating done
```
but ```make``` fails:

```
/home/apollo/Laas/pinocchio/bindings/python/algorithm/expose-model.cpp:98:27:   required from here
/usr/include/boost/variant/variant.hpp:2058:13: error: no matching function for call to ‘boost::variant<pinocchio::JointDataRevoluteTpl<double, 0, 0>, pinocchio::JointDataRevoluteTpl<double, 0, 1>, pinocchio::JointDataRevoluteTpl<double, 0, 2>, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 0> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 1> >,   [...]
usr/include/boost/variant/variant.hpp:1959:14: note: candidate: ‘template<class RhsT, class B1, class B2> [ ...] boost::detail::variant::void_; T26 = boost::detail::variant::void_; T27 = boost::detail::variant::void_; T28 = boost::detail::variant::void_; T29 = boost::detail::variant::void_]’
 1959 |         void assign_impl(
      |              ^~~~~~~~~~~
/usr/include/boost/variant/variant.hpp:1959:14: note:   template argument deduction/substitution failed:
/usr/include/boost/variant/variant.hpp:2058:13: note:   couldn’t deduce template parameter ‘B1’
 2058 |             assign_impl(
      |             ^~~~~~~~~~~
```
I understand of course that one should not expect support for such an old OS, but I would be grateful for any insight you may have.

jcarpent, 2021-08-12:
Could you send the full report? Could you use instead llvm/clang?

vincentberenz, 2021-08-12:
thanks a lot for the rapid answer.

Full report:
http://people.tuebingen.mpg.de/mpi-is-software/miscs/pinocchio_make.txt

I'll try llvm/clang

vincentberenz, 2021-08-12:
The same error occurs with clang:

http://people.tuebingen.mpg.de/mpi-is-software/miscs/pinocchio_make_clang.txt

I am afraid Max is not around these days.

jcarpent, 2021-08-12:
The issue is coming from Boost and not Pinocchio. Boost 1.54 is not totally compliant with C++11.
The thing you can do is to rely on an older version of urdfdom < 1.0.0.

This is the maximum help I can provide to you.

jcarpent, 2021-08-13:
I will close this issue. @vincentberenz Feel free to reopen it if needed.

closed 2021-08-13 by jcarpent