stack-of-tasks/pinocchio #1499: Clarify employed licenses

braunzel, 2021-08-23:
Dear Pinocchio developers, 
thank you for your incredibly valuable work. When browing through the repository, I saw that there seem to be inconsistencies in licensing. while the majority of the repo employs a permissive license, some files are explicitly marked as L-GPL. This includes
 * src/algorithm/chol.m
 * unittest/udut.cpp
 * bidings/Cmakelists.txt
 * unittest/fusion.cpp
 * doc/Overview.md
 * doc/e-implementation/c-crtp.md
 * doc/treeview.dox
Could you please clarify the licenses across the overall repo, and just in case, fix them in the individual files? 
thank you very much in advance.

jcarpent, 2021-08-23:
Thanks for this issue @braunzel. In the beginning, Pinocchio was L-GPL. 
Now, the whole project is BSD-2 (soon BSD-3 for Pinocchio 3.x)
I will fix the aforementioned files.

jcarpent, 2021-08-23:
Fix provided in #1500.

jcarpent, 2021-08-23:
Solved via #1500.

closed 2021-08-23 by jcarpent

braunzel, 2021-08-30:
//
// Copyright (c) 2015 CNRS
//
// This file is part of Pinocchio
// Pinocchio is free software: you can redistribute it
// and/or modify it under the terms of the GNU Lesser General Public
// License as published by the Free Software Foundation, either version
// 3 of the License, or (at your option) any later version.
//
// Pinocchio is distributed in the hope that it will be
// useful, but WITHOUT ANY WARRANTY; without even the implied warranty
// of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
// General Lesser Public License for more details. You should have
// received a copy of the GNU Lesser General Public License along with
// Pinocchio If not, see
// <http://www.gnu.org/licenses/>.

Still persists in src/algorithm/chol.m