stack-of-tasks/pinocchio #1515: How to obtain the Jacobian at a certain position on the manipulator link

WangXingfang, 2021-08-30:
Now, I can use pin.getJointJacobian to obtain the Jacobian of the specified joint, and use pin.getFrameJacobian to obtain the Jacobian of the specified link. But, for a certain position on the link, how can I obtain its Jacobian.

jcarpent, 2021-08-30:
At the time being, you should add a Frame to your model for that.
Soon, this won't be necessary anymore with Pinocchio 3x

closed 2021-08-30 by jcarpent

WangXingfang, 2021-08-30:
Great! Can you give me some examples or descriptions about how to add a Frame to my model?
Thanks for your fast response!

jcarpent, 2021-08-30:
`model.addFrame(Frame("name",joint_support_id,0,placement))`

WangXingfang, 2021-08-30:
Okay, thanks :)

WangXingfang, 2021-09-03:
Hello, @ jcarpent. 
While using _model.addFrame_, does _framePlacement_ depend on the _parent_idx_ coordinate system? 
And after adding a frame, how can I change its _parent_idx_ and _framePlacement_ ?

"_model.addFrame(Frame (frame_name, parent_idx, 0, framePlacement, OP_FRAME))_"

jcarpent, 2021-09-03:
The frame placement is wrt the parent joint frame.

jcarpent, 2021-09-03:
> And after adding a frame, how can I change its _parent_idx_ and _framePlacement_ ?

Why do you need to change it?
But you can do by changing the fields within the frame (model.frames[your_frame_id].placement)

WangXingfang, 2021-09-03:
> Why do you need to change it?

Because I want to get the Jacobian of a certain point on the manipulator, and the point's placement changes in real time. So  I need to change its parent and placement, do you have a better idea?

jcarpent, 2021-09-03:
> o I need to change its parent and placement, do you have a better idea?

My previous answer will be then sufficient.

WangXingfang, 2021-09-03:
Okay, I will try to solve the problem with your answer. Thanks :)

Mingrui-Yu, 2024-03-27:
> At the time being, you should add a Frame to your model for that. Soon, this won't be necessary anymore with Pinocchio 3x

May I know if the Pinocchio 3x has been released? I noticed that the latest released version seems to be 2.7.0.

nim65s, 2024-03-27:
@Mingrui-Yu not yet