stack-of-tasks/pinocchio #1516: Derivative of Buoyant Force with Respect to State

EpicDuckPotato, 2021-08-31:
Hi,
I posted an issue previously concerning the robot below.
[image]
I'm interested in adding buoyant forces to my model, where each link is subject to a constant upward force. Since the generalized buoyant force on the robot is dependent on the configuration, the forward dynamics derivatives would thereby change. Is there a convenient way to compute the forward dynamics derivatives in the presence of world-frame external forces on each link?

jcarpent, 2021-08-31:
Please provide a set of equations to get more insight ?

jcarpent, 2021-08-31:
We mostly need to know how does this external forces evolved, because I do think they are simply local forces ;)

cmastalli, 2021-08-31:
The buoyancy forces are the ones obtained using the Archimedes principles. Therefore, if my understanding is correct, these forces are applied in the CoM point of each body, which indeed could be expressed as local forces.

EpicDuckPotato, 2021-08-31:
[image]

In the image, F_b is the total buoyant force, and the summation is over links. J_l is the frame Jacobian for the center of buoyancy of link l. R_w^l rotates from the world frame into the local frame of link l. m_{l, b} is the mass of water displaced by the link, and g = -9.81. pinocchio's aba algorithm allows me to add forces in the local frame of joints, but in my case, I have a force that is always directed upward in the world frame.

jcarpent, 2021-08-31:
Yes, you can use the LOCAL_WORLD_ALIGNED coordinate system.
Also, you can easily translate the expression of your external forces to express it in the LOCAL coordinate systems.
I think you have your answer. I will then close the issue.

closed 2021-08-31 by jcarpent

EpicDuckPotato, 2021-08-31:
Just to clarify, the force is always directed upward in the world frame, so the orientation in the local frame is actually not constant, which is why I'm not sure that I can do what you described.

EpicDuckPotato, 2021-09-01:
I followed up in stack-of-tasks/pinocchio/#1521 since this issue was closed, but I guess I should just continue here. Do you have an answer my previous comment? i.e. if my forces are constant in the world frame (not local frame), is it still correct to transform them to the local frame and then compute the ABA derivatives? Because for an infinitesimal change in state, the local direction of the forces will change.

jcarpent, 2021-09-01:
You need to add to the current derivatives wrt to q the term M^-1J^T where J is the Jacobian related to the point of application of your Buoyant force.

jcarpent, 2021-09-01:
@EpicDuckPotato I do think these answers go a bit beyond what someone can expect from software developers. Isn't it?

EpicDuckPotato, 2021-09-01:
Alright yeah sorry I guess as far as understanding fundamental physics, I should probably consult elsewhere