stack-of-tasks/pinocchio #1520: Semantics of AMENT_PREFIX_PATH variable?

traversaro, 2021-08-31:
In https://github.com/stack-of-tasks/pinocchio/issues/1358 and https://github.com/stack-of-tasks/pinocchio/pull/1359, for supporting ROS2 the feature of reading from the `AMENT_PREFIX_PATH` variable was added, treating it as it contained the same directories contained in `ROS_PACKAGE_PATH` . 

However, debugging another problem, I investigated this and it seems that while `ROS_PACKAGE_PATH` contains the directories in the form `<install_prefix>/share`, `AMENT_PREFIX_PATH` instead contains directories in the form `<install_prefix>` . 

I am not sure what is the best thing to do in this case, but I wonder if the `share` subdirectory should be added to the values obtained from `AMENT_PREFIX_PATH`. 

Related issue: https://github.com/robotology/robotology-superbuild/issues/832 .

jcarpent, 2021-08-31:
@traversaro Thanks for raising this issue.
As I'm not a ROS expert, I let @wxmerkt providing his thoughts.
@wxmerkt Could you provide your feedback on this issue?

jcarpent, 2021-09-01:
I've just checked robotology/robotology-superbuild#832, and the solution sounds fine for me.
@traversaro Could you translate your change to Pinocchio?

traversaro, 2021-09-01:
Sure, let's wait to see if @wxmerkt has any feedback as I will be interested in it even for iDynTree/iCub's models. I wonder if @olivier-stasse may have any comment as he is the one that originally created https://github.com/stack-of-tasks/pinocchio/issues/1358 .

wxmerkt, 2021-09-01:
Hi @traversaro and @jcarpent,
Apologies for the slow response to this; yesterday and today are filled with meetings. I haven't used ROS2 myself and hence not tested the `AMENT_PREFIX_PATH` - but what you wrote about changing to `<install-prefix>` sounds familiar. It may be worth testing in a Docker with a commonly used/binary-available robot description package to confirm any fixes

olivier-stasse, 2021-09-01:
Dear all, the solution implemented in  robotology/robotology-superbuild#832 sounds fine for me too.

proyan, 2021-12-07:
Fixed with #1568.

closed 2021-12-07 by proyan