stack-of-tasks/pinocchio #1522: Same value for different frames when using getFrameJacobianTimeVariation

amrn9674, 2021-09-06:
Dear all,

I am using Pinocchio in my project (thanks for creating this package :+1: ), and I need to have access to the Hessian matrix. As I went through other issues, I found it is not available right now. So I decided to compute it with the help of Jacobian time Derivation. I need these derivations to be in LOCAL_WORLD_ALIGNED, however, as I tested it with different frames, it gives me the same answer, which is not the type I prefer. Thank you in advance for your help.

jcarpent, 2021-09-06:
The things is that we cannot anything with such a little amount of details and information.

amrn9674, 2021-09-06:
First of all thanks for your very quick response. Now, let me describe my project a little bit more. I want to use Crocoddyl to find an optimal controller for my robot (a very simple robot 5-link 2-D planner manipulator). For the dynamic aspect, I am using Pinocchio. In this project, not only should I found optimal input commands, but also optimal external forces. Because of this, I should know the derivative of the dynamic equation w.r.t. external forces. This means I need the Hessian matrix. Unfortunately, I could not find a way to access this, so I decided to compute it with time derivation of the Jacobian matrix (by getting the values for different unit velocities). For this, I need dJ (more specifically, data.dJ) to be defined in the LOCAL_WORLD_ALIGNED reference frame, which is not. I am trying to use getFrameJacobianTimeVariation API to solve this issue. However, as I tested it with different reference frames, they all give me the same answer. I hope it clarifies the issue, but if you need more details please let me know.

jcarpent, 2021-09-06:
Could you please look here: https://github.com/stack-of-tasks/pinocchio/issues/1513
You will have all your answers.

closed 2021-09-06 by jcarpent

amrn9674, 2021-09-07:
I actually do the same thing you mentioned at #1513, I map my external forces to local joint coordinates and use computeRNEADerivatives(model, data, q, v, a, Fext) to compute the derivatives. However, the derivatives do not agree with my analytical solution. The system is a very simple system (please see the attached picture). I test the results for when all q's, v's, and u's are zero and an external force is applied at the fifth joint (like in the picture). It means all derivatives are zero, except those which are related to the external forces, but I receive the following result for ddq_dq 
`matrix([[   0.   ,   28.547,   -3.954,    1.272,   -0.272],
        [   0.   ,  -62.403,   22.054,   -7.093,    1.515],
        [   0.   ,   38.21 ,  -32.884,   21.141,   -4.514],
        [   0.   ,   -6.459,   21.928,  -38.596,  101.893],
        [   0.   ,    2.301,   -7.812,   25.454, -148.865]])` 
which does not make sense for me as the first column is zero, and the last column is not. The forces are constant w.r.t. world frame so they do not depend on q_5, so it should be exactly the opposite way in which the last column should be zero, not the first one. My analytical solution gives 
`matrix([[   0.271,   28.818,   -3.683,    1.543,    0.   ],
        [  -1.511,  -63.914,   20.544,   -8.605,    0.   ],
        [   4.503,   42.713,  -28.382,   25.648,    0.   ],
        [-101.91 , -108.372,  -79.97 , -140.543,    0.   ],
        [ 148.807,  151.111,  140.986,  174.291,    0.   ]])`
which is more sensible for me. This problem is similar to #1222, but as I mentioned above, changing the reference frame does not change the derivative of the Jacobian matrix. I wonder if there is some thing wrong with my code or the analytical solution. 

[image]

jcarpent, 2021-09-07:
Could you share your code?

jcarpent, 2021-09-07:
Sorry, I won't be able to help you. Your code is not clear/clean enough for fast debug.

amrn9674, 2021-09-07:
Thanks for your help till now. Is there any way that I can make it more clear? My question is that why ddq_dq_analyt (in the code) is not equal to ddq_dq (calculated by Pinocchio). I think if the derivatives of J be presented in the LOCAL_WORLD_ALIGNED the problem would be solved and these two would be equal. However,  'getFrameJacobianTimeVariation' gives me the same answer noo matter which reference frame I use. I am afraid if this also happens in the background code of 'computeRNEADerivatives(hmodel, hdata, q, v, ddq,Fext)'.

jcarpent, 2021-09-08:
The thing is that you are using `getFrameJacobianTimeVariation` while you should rather use `getFrameJacobian`.

amrn9674, 2021-09-08:
But I want its derivation, not itself.

jcarpent, 2021-09-08:
But you are calling the time derivatives here.
`J_frame = dM/dq` where M is the placement of the frame.
You should first write the math.

amrn9674, 2021-09-08:
Yes, I want to calculate other derivatives with help of the time derivative. Mathematically speaking, we have 
`ddq = M^{-1} [(tau + J^T F_{ext}) -nle]`. I want to find ddq_dq, so I have to take the derivative of the right-hand side of this equation. The j-th column of ddq_dq is the derivation of ddq w.r.t. q_j which is equal to `ddq(:,j) = d(M^{-1})_dq_j [(tau + J^T F_{ext}) -nle] + M^{-1} [( (dJ_dq_j)^T F_{ext}) -dnle_dq_j]` This equation should be same for any coordinate in which F_ext defined, I mean the dependency of ddq to q_j is independent of local frame definition of F or global frame definition of it. I have the analytical solution to the global frame one, but it is not as same as the one Pinocchio gives. The result of Pinocchio for ddq_dq has some problems (I guess) which are similar to what was discussed in #1222. This means probably the derivative of Jacobian is presented in the WORLD frame, not LOCAL_WORLD_ALIGNED. Rather than whole this problem, I noticed another symptom for this which is that `getFrameJacobianTimeVariation`returns the same value for both of the reference frames. To summarize it:

1. The ddq_dq calculated with Pinocchio is not equal to what I have calculated analytically, so one of them is wrong!!! According to what I discussed in some comments above, The results of Pinocchio do not make sense. 
2. I think the problem with Pinocchio is that it calculates the derivatives in the wrong frame, or at least there is a problem with my code which it happens.
3. Another reason which somehow makes me think that there is something wrong with Pinocchio is that `getFrameJacobianTimeVariation` returns the same value for all coordinate systems, which I think it should not.

jcarpent, 2021-09-08:
Why do you use `getFrameJacobianTimeVariation`? This is time derivatives
You should use the derivatives of Pinocchio as mentioned in #1222 with external forces (expressed in the local frame of the joints).
You should also add the fact that your forces are not constant in the local frame.

amrn9674, 2021-09-08:
I did want to use time derivative to calculate the hessian like `dJ_dt = dJ_dq1 v1 + dJ_dq2 v2 + ...`. Regarding your last point, How should I consider that? Probably this is the problem.

jcarpent, 2021-09-08:
I do think you should read #1222.

amrn9674, 2021-09-08:
I think you mean #1513, don't you? I have read it for multiple times, there I think it is mentioned that the forces are constant w.r.t. the local frame, which is not the case here. However you mentioned that we should add` M^{-1}J^T` to the derivatives, do you mean this? this should be added to Fu, not Fx, am I right? Also, this issue is very related to #1516.

jcarpent, 2021-09-08:
>  However you mentioned that we should add` M^{-1}J^T` to the derivatives, do you mean this?

Yes.

amrn9674, 2021-09-08:
I do not understand this, where we should add it, to ddq_dq? Can you explain why?

jcarpent, 2021-09-08:
Do you realize that you're really demanding? Do you have a supervisor to help you?

amrn9674, 2021-09-08:
Sorry for bothering you. My problem was to understand Pinocchio. I already have the analytical solution to this problem.

jcarpent, 2021-09-08:
Please see Eq (14) here https://hal.laas.fr/hal-01790971v2/document

jcarpent, 2021-09-08:
Which means that, you need to add to your derivatives:
```
dtau\dq += J.T * (J x fext)
```
where `J` is the Jacobian of your joint frame.
J and fext should be expressed in the same coordinate system.

jcarpent, 2021-09-09:
@amrn9674 Any news on that?

amrn9674, 2021-09-10:
Sorry for my late reply, and thanks a lot for your consideration. I was not familiar with spatial algebra, so I started reading Featherston's book to understand the equations a little bit more. After learning how to apply `x` operator, I started implementing it. After testing different ideas, I found that the second J in the above equation should be changed a little. As the location of the applied forces is not changing, the linear part of it should be zero (I am using local fext, and Jacobian defined in LOCAL frame). So I can say that the equation probably should be like `dtau\dq+  = J.T * ([0;J_angular] x fext)`. I am not sure about my code structure, so this may happen because I am missing some points. I want to continue reading the book to understand how Pinocchio works, then I think I can be more confident about what I am doing, and hopefully try to bother you less.

jcarpent, 2021-09-10:
Good point: using Pinocchio without knowing what is Spatial Algrebra is indeed a bit tricky.
Enjoy your reading: it is a nice universe.

amrn9674, 2021-09-10:
Thanks a lot for your help and your motivational words.