stack-of-tasks/pinocchio #1523: extractCartesianAxis does not account for negative axis values?

braunzel, 2021-09-08:
Dear all, 
the method [extractCartesianAxis()](https://github.com/stack-of-tasks/pinocchio/blob/280005c8d99c2485ee942b6a38c4dc0edf75c706/src/parsers/urdf/model.hxx#L262)
only considers positive unit vectors, and does not account for negative axes, such as (-1, 0, 0).
URDF, by contrast, explicitly allows negative axes, http://wiki.ros.org/urdf/XML/joint, the constraint is only that the vector should be normalised.
That means, possibly lots of joints get added as "unaligned". Does that create any efficiency difference?
Thanks in advance.

jcarpent, 2021-09-08:
You're totally correct. But I would say even Joint{Revolute,Prismatic}Unaligned are already extremely optimized.

We may think to either:
- add Joint{Revolute,Prismatic}{X,Y,Z}Minus, but at the price of increase of compilation timings
- handle this context by applying a Rotation which the joint axis into its opposite.
- let's at it is now.

@jmirabel @nim65s @proyan @nmansard @traversaro @wxmerkt Do you have any opinion on that?

jcarpent, 2021-09-08:
I'm mostly in favor of option 2., which is fast to implement.

jmirabel, 2021-09-09:
I would rather be in favor of a runtime solution rather than a compile-time one (like having a boolean in JointRX that tells you whether you should use `theta` or `-theta`. Not sure if this breaks the API.

If you go for 2, you should factorize the code of the positive and negative axis in the same template class.

braunzel, 2021-09-09:
Also, maybe consider preferring the notation `Vector3::UnitX()` over `(1., 0., 0.)`.

traversaro, 2021-09-09:
> 2. handle this context by applying a Rotation which the joint axis into its opposite.

I am not sure I got all the details of this proposal, but if the idea is to change the link frames to be aligned with the joint axis, I think it would be ideal if it could be made explicit (for example with an explicit function that process the model and return a new model in output), so that users interested in squeezing all the performance can opt in, while users that are interested in keeping the link frames expressed as in the URDF do not have their link frames silently changed.

jmirabel, 2021-09-09:
@traversaro proposal is better than mine.

traversaro, 2021-09-09:
> I guess @jcarpent meant to modify the `jointPlacements` attribute, all frame placements attached to this joint and the inertia.

I see, then I think I got it. I wonder if it could make sense to have a `buildOptimizedModel` (name bike-shedding would be welcome : ) ) similar to `buildReducedModel` .

jmirabel, 2021-09-09:
Yep, and it could replace all unaligned joints.

proyan, 2021-09-09:
@jmirabel's suggestion about a simple boolean should be easy to implement, but would add conditional branches at a very low level for cppad and casadi scalars. I don't think they would like that.

Once the model is parsed, is there any advantage to keeping the original minus sign for the axis, or the original axis information? I don't see any, but perhaps I'm missing something. 

We already have been discussing something similar to `buildOptimizedModel` for creating more kinematicaly efficient chains from the original model. I think @traversaro's proposal could fit in that, without any other separate API. What do you think?

proyan, 2021-12-06:
This issue would be handled along with the TransmissionModel in pinocchio3.