stack-of-tasks/pinocchio #1525: Jacobian calculation in Local and World

FirasAbiFarraj, 2021-09-10:
Hello,

I was trying to estimate the jacobian of the end effector with respect to the base frame. I thought doing things in these two ways should give the same result but it does not really give the same result. In fact, i am lost of what the jacobian calculated in the _World_ reference is.

Method 1 (not giving what I expect):
```
pinocchio::computeFrameJacobian(Model, Data, q, eeFrameId_,pinocchio::ReferenceFrame::World, J);
```

Method 2  (giving what I expect):
```
pinocchio::computeFrameJacobian(Model, Data, q, eeFrameId_,pinocchio::ReferenceFrame::LOCAL, J);
pinocchio::framesForwardKinematics(Model, Data, q);
Eigen::MatrixXd rotateJacobian(6,6);
rotateJacobian.setZero();
rotateJacobian.block(0,0,3,3) = Data.oMf[eeFrameId_].rotation();
rotateJacobian.block(3,3,3,3) = Data.oMf[eeFrameId_].rotation();
J = rotateJacobian * J;
```

I did not really understand what this explanation of the what the world reference is means:
```
The WORLD frame convention corresponds to the frame concident with the Universe/Inertial frame but moving with the moving part (Joint, Frame, etc.). 
```

Am I missing something? Or is the jacobian in the _World_ reference frame not what I expect it to be?

FirasAbiFarraj, 2021-09-11:
Sorry if the question is redundant. Maybe I haven't read all previous issues (I read quite a bit though). Of course I have seen the LOCAL_WORLD_ALIGNED and I saw it in the documentation too. 

My question was more because I am interested in understanding what the jacobian calculated in WORLD means as it seems to be some new concept. Like why would somebody use it? 
I saw the reference to  Featherstone's book too but I don't have access to it now so I thought I'd ask. It is not so important as I am using the library and it works perfectly (thanks for doing this great job !). It was more out of curiosity.

jcarpent, 2021-09-11:
The WORLD coordinate system refers to the origin of the world. So when you expressed a velocity of a joint in the WORLD, it refers to the velocity of the point coincident with the origin of the world, but attached to the joint in motion. This coordinate system is not useful in general, except to enhance the computations.

FirasAbiFarraj, 2021-09-13:
Perfect ! Thank you !