stack-of-tasks/pinocchio #1526: Question about using analytical derivatives for contact

Kongx231, 2021-09-10:
I'm a little confused where the contact Jacobian J comes in when calling computeRNEADerivatives or computeABADerivatives for constrained modes. I know the f_ext contains the reaction forces, but I'm not sure where how it has information about the contact. Is there a barebone example (preferably in python)? I see the unit test contact-dynamics-derivatives.cpp computes the derivative, and I'm assuming it is in the data_check struct but its not entirely clear to me.

Kongx231, 2021-09-14:
Thanks! I see computeConstraintDynamicsDerivatives which contains the contact model. Are there python wrappers for these functions/objects?

jcarpent, 2021-09-14:
Yes, of course.

chenxiaocongAI, 2025-09-13:
@Kongx231  hi brother, have you completed the dynamics to make contact estimage?