stack-of-tasks/pinocchio #1528: No module named 'pinocchio.pinocchio_pywrap'

Kongx231, 2021-09-15:
I've built and installed pinocchio through the installation procedure, but I'm having trouble importing pinocchio from the build

From make install I get that pinocchio_pywrap is installed:

-- Set runtime path of "/usr/local/lib/python3/dist-packages/pinocchio/pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so.2.6.3" to ""
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/__init__.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/deprecated.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/deprecation.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/utils.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/robot_wrapper.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/romeo_wrapper.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/explog.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/shortcuts.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/visualize/__init__.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/visualize/base_visualizer.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/visualize/gepetto_visualizer.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/visualize/meshcat_visualizer.py
-- Installing: /usr/local/lib/python3/dist-packages/pinocchio/visualize/panda3d_visualizer.py

when I look in /usr/local/lib/python3/dist-packages/pinocchio, I see that pinocchio_pywrap is in there
deprecated.py
deprecation.py
explog.py
__init__.py
pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so
pinocchio_pywrap.cpython-36m-x86_64-linux-gnu.so.2.6.3
pinocchio_pywrap_default.cpython-36m-x86_64-linux-gnu.so
pinocchio_pywrap_default.cpython-36m-x86_64-linux-gnu.so.2.9.1
robot_wrapper.py
romeo_wrapper.py
shortcuts.py
utils.py
visualize

And my path contains 
 '/opt/openrobots/lib/python3.6/site-packages', '/usr/local/lib/python3/dist-packages/pinocchio', '/usr/local/lib/python3/dist-packages'

However, when I try to import pinocchio, I get the following error:

jcarpent, 2021-09-16:
Please remove all the Pinocchio path within your site packages folder. It seems you have a very dirty installation. Otherwise use pip or conda to get a fresh and clean install.

closed 2021-09-16 by jcarpent

nim65s, 2021-09-16:
Your setup from source in `/usr/local` and your setup from robotpkg binaries in `/opt/openrobots` are for the python interpreter of your system, which is python 3.6. The error you show at the end is a python 3.8 from conda. 

Conda is not compatible with anything else: please use only conda, or don't use conda at all.

Also, in your path, `/usr/local/lib/python3/dist-packages/pinocchio` is wrong: you should remove it.

Kongx231, 2021-09-16:
Making those changes fixed the issue. Thank you!