stack-of-tasks/pinocchio #1529: Feature proposal: Batch Computation of FK, Jacobian ...

TheCamusean, 2021-09-17:
Hello,

I was wondering if Pinocchio provides the option of batch computation of the robot kinematics and dynamics. In sample based optimization problems, it might be interesting to evaluate multiple possible robot configurations in batch. 

Is this property already implemented? In case is not; are there future plans of adding batch computation properties? I guess it could be done by modifying the data variable to a list of data variables.

Best regards,
-Julen

jcarpent, 2021-09-17:
Yes, we have this support for some algos. The principles have been implemented with the notion of ModelPool (https://github.com/stack-of-tasks/pinocchio/blob/master/src/multibody/pool/model.hpp).
Please see here for further details: https://github.com/stack-of-tasks/pinocchio/tree/master/src/algorithm/parallel

I did not have time to parallelized the evaluation for all the algorithms, but this is not too hard to extend.

closed 2021-09-17 by jcarpent

TheCamusean, 2021-09-17:
Thanks alot! :)

jcarpent, 2021-09-17:
If you think a signature is missing, do not hesitate to provide it with a PR.