stack-of-tasks/pinocchio #1531: Using constraint dynamics within python

Kongx231, 2021-09-17:
I had a couple of questions about using functions like impulseDynamics, computeImpulseDynamicsDerivatives, constraintDynamics, and computeConstraintDynamiscDerivatives in python. Specifically, how do you initialize the contact model? And do you have any tips on getting these functions to work in python?

I know in C++ you use

`  const PINOCCHIO_STD_VECTOR_WITH_EIGEN_ALLOCATOR(RigidConstraintModel) empty_contact_models;
  PINOCCHIO_STD_VECTOR_WITH_EIGEN_ALLOCATOR(RigidConstraintData) empty_contact_data;
initConstraintDynamics(model,data,empty_contact_models);
`

Also, how come you don't need to define the contact Jacobian like you do when you call forwardDynamics?

proyan, 2021-09-17:
Hi Kongx231

You can see the use of constraint dynamics in examples at https://github.com/proyan/pinocchio/blob/pinocchio3-preview-ci/examples/simulation-contact-dynamics.py

Regarding the use of contact jacobian, I recommend you to read the associated paper https://hal.inria.fr/hal-03271811/file/rss-proximal-and-sparse.pdf to understand the proposed API

Kongx231, 2021-09-17:
Thanks! I think the example will help a lot.

Kongx231, 2021-09-17:
The example works for me. However, when I try to compute the derivatives, I get a pywrap error saying the signatures don't match. I also cannot get the impulse to work either. I've looked at the source files, and I feel like this is the correct input for them especially since the constraint dynamics works. Do you have any suggestions?

I've modified the code to compute the derivatives after I compute the constraint dynamics. 

`prox_settings = pin.ProximalSettings(1e-12,1e-12,10)
    # data_sim.M[np.tril_indices(np.shape(data_sim.M)[0],np.shape(data_sim.M)[1])] = (np.triu(data_sim.M).T).flatten()

    a = pin.constraintDynamics(model,data_sim,q,v,tau,contact_models,contact_datas,prox_settings)

    # Set the lower right triangle equal to the upper right triangle (data.M.triangularView<Eigen::StrictlyLower>()= data.M.transpose().triangularView<Eigen::StrictlyLower>();)
    data_sim.M[np.tril_indices(np.shape(data_sim.M)[0])] = data_sim.M.T[np.tril_indices(np.shape(data_sim.M)[0])]

    # Try constraint derivative
    pin.computeConstraintDynamicsDerivatives(model, data_sim,contact_models ,contact_datas)

    # Try impulse
    r_coeff = 0
    mu0  = 0
    pin.impulseDynamics(model, data_sim,contact_models,contact_datas, r_coeff, mu0)

    # Try impulse derivatives
    pin.computeImpulseDynamicsDerivatives(model, data_sim,contact_models,contact_datas, r_coeff,mu0)`

The error for the constraint derivatives is 

Traceback (most recent call last):
  File "/home/nathankong/pinocchio3/examples/simulation-contact-dynamics.py", line 120, in <module>
    pin.computeConstraintDynamicsDerivatives(model, data_sim,contact_models ,contact_datas)
Boost.Python.ArgumentError: Python argument types in
    pinocchio.pinocchio_pywrap_default.computeConstraintDynamicsDerivatives(Model, Data, list, list)
did not match C++ signature:
    computeConstraintDynamicsDerivatives(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas)

Process finished with exit code 1

And then if I comment out the derivative part and try to do the impulse dynamics, I also get this error

Traceback (most recent call last):
  File "/home/nathankong/pinocchio3/examples/simulation-contact-dynamics.py", line 125, in <module>
    pin.impulseDynamics(model, data_sim,contact_models,contact_datas, r_coeff, mu0)
  File "/usr/local/lib/python3/dist-packages/pinocchio/deprecated.py", line 32, in impulseDynamics
    return pin.impulseDynamics(model, data, *args)
Boost.Python.ArgumentError: Python argument types in
    pinocchio.pinocchio_pywrap_default.impulseDynamics(Model, Data, list, list, int, int)
did not match C++ signature:
    impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas)
    impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas, double r_coeff)
    impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas, double r_coeff, double mu)
    impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian)
    impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient)
[... cut ...] atrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient)
    impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient, double damping)

Process finished with exit code 1

jcarpent, 2021-09-18:
It is just a matter of adding
```c++
#include "pinocchio/bindings/python/utils/std-vector.hpp"
```
at the top of the files exposing the impulseDynamics or derivatives functions.

closed 2021-09-18 by jcarpent

Kongx231, 2021-09-18:
The constrained derivatives file was missing 

`#include "pinocchio/bindings/python/utils/std-vector.hpp"`

and adding it in fixed the issue for that file. However, the impulse dynamics files already have that include in and still returns the same error as before.

Traceback (most recent call last):
File "/home/nathankong/pinocchio3/examples/simulation-contact-dynamics.py", line 125, in
pin.impulseDynamics(model, data_sim,contact_models,contact_datas, r_coeff, mu0)
File "/usr/local/lib/python3/dist-packages/pinocchio/deprecated.py", line 32, in impulseDynamics
return pin.impulseDynamics(model, data, *args)
Boost.Python.ArgumentError: Python argument types in
pinocchio.pinocchio_pywrap_default.impulseDynamics(Model, Data, list, list, int, int)
did not match C++ signature:
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas, double r_coeff)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v, std::vector<pinocchio::RigidConstraintModelTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintModelTpl<double, 0> > > contact_models, std::vector<pinocchio::RigidConstraintDataTpl<double, 0>, Eigen::aligned_allocator<pinocchio::RigidConstraintDataTpl<double, 0> > > {lvalue} contact_datas, double r_coeff, double mu)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient, double damping)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient)
impulseDynamics(pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> model, pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl> {lvalue} data, Eigen::Matrix<double, -1, 1, 0, -1, 1> q, Eigen::Matrix<double, -1, 1, 0, -1, 1> v_before, Eigen::Matrix<double, -1, -1, 0, -1, -1> constraint_jacobian, double restitution_coefficient, double damping)

Process finished with exit code 1

jcarpent, 2021-09-18:
You're missing the config and velocity vectors

Kongx231, 2021-09-18:
Great, thanks!