stack-of-tasks/pinocchio #1534: Joint accelerations using forward dynamics seem large

Kongx231, 2021-10-05:
I'm trying to get the forward dynamics for walking, and I have been using constraintDynamics. However, the joint accelerations seem rather large for the swing legs. When I compare the joint accelerations to other solvers, they don't seem to match well. 

I've simplified the problem to calling aba and having just a pendulum with length=2, mass=1, inertia=1, g=9.81, center of mass located at 1 and swinging about its edge. When I initialize the pendulum at pi/2 with no velocity, the acceleration should be around -4.905 but I am getting around -7.7. 

I'm not sure what the cause of the higher joint accelerations are. Any help would be great.

I'm running the pinocchio3-preview and I've attached a tar file with my python code and urdf if that helps.
[simple_pendulum.tar.gz](https://github.com/stack-of-tasks/pinocchio/files/7288216/simple_pendulum.tar.gz)

jcarpent, 2021-10-05:
Dear @Kongx231,

First of all, thanks for the clean code.
It appears you have forgotten to normalize the configuration vector as follows:
```
import sys, os
import pinocchio as pin
from os.path import dirname, join, abspath
import numpy as np

pinocchio_model_dir = join(dirname(dirname(str(abspath(__file__)))), "simple_pendulum")

print(pinocchio_model_dir)
mesh_dir = pinocchio_model_dir

urdf_filename = "/single_pendulum.urdf"
urdf_model_path = pinocchio_model_dir + urdf_filename

model, collision_model, visual_model = pin.buildModelsFromUrdf(urdf_model_path,mesh_dir)
data = model.createData()

q = np.array([1, np.pi/2])
q = pin.normalize(q)
v = np.array([0])
tau = np.array([0])
a = pin.aba(model,data,q,v,tau)
print(a)
```

closed 2021-10-05 by jcarpent

junlileeds, 2021-11-19:
Dear @jcarpent ,

There is no complex joints such as floating-base joint or continuous joint in the pendulum system, why we still need to normalize the configuration vector before calling forward dynamics algorithms? 

Thanks.

jcarpent, 2021-11-19:
Because it is using continuous joints formulation, parametrized by q = [cos(theta),sin(theta)].

junlileeds, 2021-11-19:
so sorry for disturbing.

jmirabel, 2021-11-19:
For others bumping into this, it should be
`q = pin.normalize(model, q)`.