stack-of-tasks/pinocchio #1535: I cannot find "computeJointJacobiansTimeVariation" in robot_wrapper.py

JhonPool4, 2021-10-18:
Hi, I am using Ubuntu 20.04 with python3.8. I followed the installation instructions of [link](https://stack-of-tasks.github.io/pinocchio/download.html) and just modify the installation line as  `sudo apt install -qqy robotpkg-py38-pinocchio`. After the installation, I configured the environment variables. Now, I can load a file .urdf, compute inertia matrix and jacobian, but when trying to compute the derivative in time of jacobian matrix with ` pin.computeJointJacobiansTimeVariation(robot.model, robot.data, q0, dq0)` I get this error: `AttributeError: 'RobotWrapper' object has no attribute 'computeJointJacobiansTimeVariation'`. So I checked the "robot_wrapper.py" file and realized that the function was not there. Please, could you tell me how to get the function? 
Thanks,

jcarpent, 2021-10-18:
`pin.computeJointJacobiansTimeVariation(robot.model, robot.data, q0, dq0)` should then work smoothly.

jcarpent, 2021-10-18:
Currently, we did not expose the method yet. If you think this feature is missing, you can easily collaborate on the project by adding it.