stack-of-tasks/pinocchio #1543: Converting force vectors to local frame

manumerous, 2021-11-01:
Hi I am trying to call this [inverse dynamics rnea function ](https://github.com/stack-of-tasks/pinocchio/blob/5daf5de1c158363e835d322435bab4aa2f9b9b0b/src/algorithm/rnea.hpp#L55) with some external forces. 

Does the pipeline already contain functionalities to convert a vector from the world frame to the local frames of the joints using a joint name or joint ID? 

So far I could not find it.

jcarpent, 2021-11-01:
Not yet. You should do that manually. This feature will be present in Pinocchio 3

closed 2021-11-01 by jcarpent

manumerous, 2021-11-01:
Okay thanks a lot for the fast feedback! I am in need of that feature for a project of mine, so I will implement it manually then.