stack-of-tasks/pinocchio #1548: Inertia Matrix derivative

bhtxy0525, 2021-11-05:
Is there a function in Pinocchio that supports taking derivatives of the inertia matrices?
I want to verify whether M-2*C satisfies the anti-symmetric property, which is an important characteristic in robot dynamics control

yangyanboxiaoyao, 2021-11-05:
I have the same question about the Coriolis matrix

jcarpent, 2021-11-05:
Thanks for raising this interesting question.
@bhtxy0525 I guess you meant `dM\dt - 2*C` should anti-symmetric. 
@yangyanboxiaoyao you can use `computeCoriolisMatrix` to efficiently compute C

Concerning both concerns, here is a script to compute an estimate of the `dM/dt`
```python
import pinocchio as pin
import numpy as np
import sys
import os
from os.path import dirname, join, abspath

from pinocchio.visualize import MeshcatVisualizer

# Load the URDF model.
# Conversion with str seems to be necessary when executing this file with ipython
pinocchio_model_dir = join(dirname(dirname(str(abspath(__file__)))),"models")

model_path = join(pinocchio_model_dir,"example-robot-data/robots")
mesh_dir = pinocchio_model_dir
urdf_filename = "icub.urdf"
urdf_model_path = join(join(model_path,"icub_description/robots"),urdf_filename)

model, collision_model, visual_model = pin.buildModelsFromUrdf(urdf_model_path, mesh_dir, pin.JointModelFreeFlyer())

# Display a robot configuration.
q0 = pin.neutral(model)
v0 = np.random.rand(model.nv)

data = model.createData()
M = pin.crba(model,data,q0)

dt = 1e-8
q_plus = pin.integrate(model,q0,v0*dt)

data_plus = model.createData()
M_plus = pin.crba(model,data_plus,q_plus)

dM = (M_plus - M)/dt

C = pin.computeCoriolisMatrix(model,data,q0,v0)

A = dM - 2*C
res = A + A.T

assert np.allclose(res,np.zeros((model.nv,model.nv)),atol=1e-6)

```

@bhtxy0525 Do you really need an algorithm to get `dM/dt`? I may take some to add it within Pinocchio but I do not think this one is essential.

bhtxy0525, 2021-11-05:
@jcarpent  thanks you so much.
I  have run your recommended program and the result showed that dM/dt - 2*C satisfies the anti-symmetric property. So i guess the C realized by **computeCoriolisMatrix**   satisfies the first kind of Christoffel, am I right? 
Finally, i don't think there is an independent algorithm to get dM/dt, thank you again.

jcarpent, 2021-11-05:
> So i guess the C realized by **computeCoriolisMatrix** satisfies the first kind of Christoffel, am I right?

Yes, you're totally right.

closed 2021-11-05 by jcarpent

bhtxy0525, 2021-11-05:
@jcarpent Thank you for taking the time to reply.

yangyanboxiaoyao, 2021-11-05:
@jcarpent ,thank you for your reply, could you tell me which paper or textbook method is used to calculate function `computeCoriolisMatrix ` ?

jcarpent, 2021-11-05:
I wrote this algorithm on my own, following the development of the analytical derivatives (https://hal.archives-ouvertes.fr/hal-01790971). It was already 4 years ago.
I did not take the time to write a proper document on it, but if you think it is worth it, I may spend some time. It is just I have limited working bandwidth at this time.

Thanks for considering Pinocchio. Any feedback is much more than welcome.

pwensing, 2022-06-06:
@bhtxy0525  @yangyanboxiaoyao In case this topic is still of interest, please see an updated resolution in discussion #1663  and associated changes in #1665. Many thanks to @jcarpent for the recent updates.

bhtxy0525, 2022-06-06:
@pwensing thanks you very much! I have read discussion #1663  and noticed that a matrix C satisfying the skew-symmetric property is not unique. Indeed, the previous C obtained by Pinnochio is inconsistent with the C calculated using MATLAB （satisfied the Christoffel symbol of the first kind ）. Now, i can't wait to verify this new Coriolis matrix modified by @jcarpent .
Thank you all ！！！

jcarpent, 2022-06-06:
@bhtxy0525 You can access it on the devel branch (install from source).