stack-of-tasks/pinocchio #1550: Problem accessing `geometryObjects` in Python on MacOS.

duburcqa, 2021-11-11:
I'm compiling pinocchio on MacOS. Almost everything works fine excepts on small bug that is unfortunately blocking for me. I cannot access methods and properties of derived primitive shapes. For instance if I do this:
```
import pinocchio

model = pinocchio.buildModelFromUrdf("./data/sphere_primitive.urdf")
collision_model = pinocchio.buildGeomFromUrdf(model, "./data/sphere_primitive.urdf", pinocchio.COLLISION)
collision_model.geometryObjects[0].geometry.clone()
```
I get this error:
```
ArgumentError: Python argument types in
    Sphere.clone(Sphere)
did not match C++ signature:
    clone(hpp::fcl::Sphere {lvalue})
```

Note that if I create primitive shapes like `Sphere` directly using `hppfcl` bindings, I can access all its proprieties as expected. Moreover, everything is working fine both on linux and windows. All the dependencies are built statically except boost python itself to take advantage of the shared register in order to avoid this kind of issues in the first place. 

For reproducibility, the content of the urdf file "./data/sphere_primitive.urdf" is this:
```
<?xml version="1.0" ?>
<robot name="sphere">
    <link name="MassBody">
        <inertial>
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
            <mass value="1.0"/>
            <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
        </inertial>
        <visual>
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
            <geometry>
                <sphere radius="0.5"/>
            </geometry>
            <material name="">
                <color rgba="0.65 0.65 0.65 1.0"/>
            </material>
        </visual>
        <collision>
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
            <geometry>
                <sphere radius="0.5"/>
            </geometry>
        </collision>
    </link>
</robot>
```

jcarpent, 2021-11-11:
With Pinocchio installed with conda, it works totally fine.
I suspect this is coming from the way you installed Pinocchio+HPP-FCL on OS X.

duburcqa, 2021-11-11:
Hum... ok I'm not really surprise. Thank you for checking!

duburcqa, 2021-11-11:
Just for you to know, I can confirm 100% that it comes from the fact that I statically build hpp-fcl.

closed 2021-11-11 by duburcqa