stack-of-tasks/pinocchio #1553: Closed loop dynamics constraints with Pinocchio preview 3

wchengh2010, 2021-11-15:
Hi,

Thank you for maintaining this very well known repo. It has been greatly helpful for our research work. We are working on Cassie and Digit robot which have closed loop 4-bar linkage. We have been using RBDL with closed loop dynamics and want to switch to Pinocchio preview 3. We have the following 3 questions. Thanks a lot for taking your time to answer our questions!

1. In RBDL, there is 'AddLoopConstraint' function to add closed loop constraints with predecessor and successor bodies. Do we have a similar function in pinocchio preview 3? I believe it's 'RigidConstraintModel', but do we have an API interface explanation on how to use it? 
2. About importing close loop constraint dynamics, are we supporting it in the parser? If we have an sdf file with close loop linkage, will the preview 3 parser be able to parse it and do we have an added joint type for handling the closed loop constraints?
3. I checked out the following unit test as an example for the closed loop dynamics: 
https://github.com/stack-of-tasks/pinocchio/blob/pinocchio3-preview/unittest/closed-loop-dynamics.cpp
Based on my understanding, the process to solve closed loop dynamics is to 1. Use 'RigidConstraintModel' to add constraints to the 'contact_models' and 'contact_data'. 2. Call 'initConstraintDynamics' and then 'constraintDynamics' to solve forward dynamics. To get derivatives, are we using 'computeConstraintDynamicsDerivatives' function? 

Thank you very much!

jcarpent, 2021-11-15:
>     1. In RBDL, there is 'AddLoopConstraint' function to add closed loop constraints with predecessor and successor bodies. Do we have a similar function in pinocchio preview 3?

The current API for closed loops handling is not fixed yet: we will provide some updates to make it more compliant and easy to use.

>      I believe it's 'RigidConstraintModel', but do we have an API interface explanation on how to use it?

You're totally correct. We will release soon the documentation of all these features. In the meantime, the example https://github.com/stack-of-tasks/pinocchio/blob/74c574f1e6b9cc1d9eb417855a6112d56a981146/examples/simulation-contact-dynamics.py should help you.

jcarpent, 2021-11-15:
>     2\. About importing close loop constraint dynamics, are we supporting it in the parser? If we have an sdf file with close loop linkage, will the preview 3 parser be able to parse it and do we have an added joint type for handling the closed loop constraints?

We are supporting SDF parsing. You can compile these parser using cmake.
Closed-loop constraints for Cassie for instance are handled via 6d constraint. While the computational is a bit increased, the advantage is that we kept the entire kinematic tree in the Model.

jcarpent, 2021-11-15:
>     3\. I checked out the following unit test as an example for the closed loop dynamics:
>         https://github.com/stack-of-tasks/pinocchio/blob/pinocchio3-preview/unittest/closed-loop-dynamics.cpp
>         Based on my understanding, the process to solve closed loop dynamics is to 1. Use 'RigidConstraintModel' to add constraints to the 'contact_models' and 'contact_data'. 2. Call 'initConstraintDynamics' and 'constraintDynamics' to solve forward dynamics. To get derivatives, are we using 'computeConstraintDynamicsDerivatives' function?

You are totally right. It seems you have completely understood the current API for closed kinematic chains and contact constraints.

jcarpent, 2021-11-15:
>  We are working on Cassie and Digit robot which have closed loop 4-bar linkage. We have been using RBDL with closed loop dynamics and want to switch to Pinocchio preview 3.

By the way, to which research team do you belong to?

NmBoyd, 2021-11-15:
> > We are working on Cassie and Digit robot which have closed loop 4-bar linkage. We have been using RBDL with closed loop dynamics and want to switch to Pinocchio preview 3.
> 
> By the way, to which research team do you belong to?

@wchengh2010 works with me at Georgia Tech on Cassie and an incoming Digit at LIDAR. Currently we are refactoring a lot of the software tools we are using. Hence the questions on the progress of closed loop kinematics. Digit atm also does not have the most put-together API at the low-level and does not provide much for dynamics/kinematics or control unfortunately.

closed 2021-11-16 by jcarpent

zapplelove, 2022-06-08:
It's a amazing work, and I hope the version 3 can be released soon. And I hope the docs can be more detailed and provide some examples using sdf parser.
Best wishes!