stack-of-tasks/pinocchio #1555: ImportError: cannot import name 'casadi' from 'pinocchio'

kploeger, 2021-11-24:
Dear all,

I would like to use Pinocchio preview 3 together with Casadi in Python to do some trajectory optimization.

Since https://github.com/conda-forge/pinocchio-feedstock#installing-pinocchio installed version 2.6.4, I tried `conda install pinocchio -c conda-forge/label/main` in a clean Python 3.8 conda environment on a fresh Ubuntu 20.04 which according to `conda list` successfully installed version 2.9.1.

Now `import pinocchio` works as expected but when I try `from pinocchio import casadi as cpin` as in https://github.com/stack-of-tasks/pinocchio/blob/pinocchio3-preview/unittest/python/casadi/bindings_main_algo.py, I get `ImportError: cannot import name 'casadi' from 'pinocchio'`. Same thing for Python 3.6

What might be the problem here?
Many thanks in advance.

jcarpent, 2021-11-24:
You should install it manually sorry.
2.9.x should not have been published so far. We were not ready to support it via.

But manual install is really easy from the pinocchio3-preview branch.
Where are you working? And how do you know this extended feature of Pinocchio?

kploeger, 2021-11-24:
Ahh okay, then I'll try the manual install.

I read somewhere, that Pinocchio works with CasADi, but didn't realize that it was only referring to C++. Then I randomly stumbled over the release notes of 2.9.0 https://github.com/stack-of-tasks/pinocchio/releases/tag/v2.9.0.

jcarpent, 2021-11-24:
It does work well. Many people around us are using it actually.

closed 2021-11-24 by jcarpent