stack-of-tasks/pinocchio #1556: Unilateral contact dynamics for MuJoCo-esque locomotion experiments

mhr, 2021-11-27:
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_b-examples_k-qp.html

Hello! I was hoping as I was reading through the documentation to find an example of how to use Pinocchio with unilateral contacts so I could do some experiments similar to the MuJoCo locomotion tasks. However, it seems such an example has not been implemented yet. Would it be possible to provide an example for such a use-case? It isn't immediately clear to me at a glance in the documentation how to accomplish this, as a beginner to Pinocchio.

proyan, 2021-11-29:
Hi @mhr 
Contacts and their derivatives are a new feature that are currently in the (still unstable) branch pinocchio3-preview. You can see the example of contact simulation here: https://github.com/stack-of-tasks/pinocchio/blob/pinocchio3-preview/examples/simulation-contact-dynamics.py

The derivatives of the contacts, similarly, can be computed simply by the `computeConstraintDynamiscDerivatives(model, data, contact_models, contact_datas, prox_settings)`. The description for contact_models and contact_datas are available in the example above, and for prox_settings I would recommend that you read the section III of the associated paper: https://hal.inria.fr/hal-03271811/file/rss-proximal-and-sparse.pdf

jcarpent, 2021-11-29:
@mhr Thanks for your interest in using Pinocchio.
We will provide soon the whole support for unilateral contact models. It is mostly a matter of limited work bandwidth.
Currently, as mentioned by @proyan, we only support bilateral contact models and their derivatives.