stack-of-tasks/pinocchio #1557: wrong generation of config cmake files

xeonz1, 2021-11-29:
pinocchioConfig.cmake

windows 10, MinGW gcc

the lib file name should be *pinocchio.lib.a instead of *pinocchio.lib
```

# Find absolute library paths for all _PKG_CONFIG_LIBS as CMake expects full paths, while pkg-config does not.
set(_PACKAGE_CONFIG_LIBRARIES "")
set("_pinocchio_PKG_CONFIG_LIBDIR" "C:/Program Files (x86)/pinocchio/lib")
set("_pinocchio_PKG_CONFIG_LIBS_LIST" " **-lpinocchio.lib** -LC:/Users/seaso/Apps/boost_1_77_0/stage/lib -llibboost_filesystem-mgw8-mt-x64-1_77.a -llibboost_serialization-mgw8-mt-x64-1_77.a -llibboost_system-mgw8-mt-x64-1_77.a -llibboost_unit_test_framework-mgw8-mt-x64-1_77.a")
```

xeonz1, 2021-11-29:
> 

Most likely because it was compiled by mingw.

jcarpent, 2021-11-29:
We may add a specific case for MinGW then. Do you really need to use MinGC?

xeonz1, 2021-11-29:
> We may add a specific case for MinGW then. Do you really need to use MinGW?

I am not sure.... I need to use C++/C on windows for the sake of communication with simulink. Personally I prefer mingw rather than MSVC. So far I dont think preparation for development using MSVC is easier than mingw...
 Any better suggestions on using pinocchio with C++? Currently I am stuck in the generation of crocoddyl makefiles, which threw a error CMake generation problem.

Already thinking about using python now....

jcarpent, 2021-11-29:
Croco is not yet compatible with Windows. It is not exporting the symbols.
I would say for Pinocchio, cmake tool is very intuitive and easy to use. Please check with the Windows CI (https://github.com/stack-of-tasks/pinocchio/blob/master/.github/workflows/windows-conda-v142.yml).

proyan, 2021-11-29:
Related to https://github.com/loco-3d/crocoddyl/issues/1014

jcarpent, 2021-11-29:
@xeonz1 If you just need to install Pinocchio, then relying on conda for that is the easy way.

xeonz1, 2021-11-30:
> @xeonz1 If you just need to install Pinocchio, then relying on conda for that is the easy way.

Not familiar with conda. Can I still use c++ with conda ver. pinocchio?

jcarpent, 2021-11-30:
> (Issue #1557)

Yes. Of course !

closed 2021-11-30 by jcarpent