stack-of-tasks/pinocchio #1558: cppadcg_eigen.hpp missing

shubhamsingh91, 2021-11-29:
When I try to compile the timings-cg.cpp in benchmark to test the codegen algos, then I get the following error. 

fatal error: cppad/cg/support/cppadcg_eigen.hpp: No such file or directory
 #include <cppad/cg/support/cppadcg_eigen.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Does building Pinocchio doesn't build CppADCodeGen automatically?

Thanks

proyan, 2021-11-29:
Did you turn ON `BUILD_WITH_AUTODIFF_SUPPORT` (for cppad) and `BUILD_WITH_CODEGEN_SUPPORT` (for cppadcodegen) when doing cmake?

shubhamsingh91, 2021-11-29:
Thanks for the reply.
Is it as simple as -BUILD_WITH_AUTODIFF_SUPPORT=ON -BUILD_WITH_CODEGEN_SUPPORT=ON added to the cmake command?

shubhamsingh91, 2021-11-29:
Figured it out!

For anyone else stuck on this-

1. I changed the flags BUILD_WITH_AUTODIFF_SUPPORT and BUILD_WITH_CODEGEN_SUPPORT in the CMakeLists to ON, 
2. Built cppAD from https://coin-or.github.io/CppAD/doc/install.htm using cmake.
3. Built cppADcg from https://github.com/joaoleal/CppADCodeGen using cmake.
4. compiled the Pinocchio source file.

closed 2021-11-29 by shubhamsingh91