stack-of-tasks/pinocchio #1559: Confliction when building on Windows

xeonz1, 2021-12-01:
In pinocchio_math_fwd_hpp,

it defines :
PINOCCHIO_OVERLOAD_MATH_BINARY_OPERATOR(min)

pinocchio/fwd.hpp includes windows.h, which includes winddef, and then minwindef. This leads to confliction.

Using conda version pinocchio for c++ building currently.

cmakelists:

`cmake_minimum_required(VERSION 3.0.0)
project(pytest)

find_package(pinocchio REQUIRED)
# find_package(Eigen3 REQUIRED)
# find_package(casadi REQUIRED)

set (CMAKE_CXX_STANDARD 17)
add_executable(pytest main.cpp)

target_link_libraries(pytest PUBLIC pinocchio::pinocchio)
`

jcarpent, 2021-12-01:
I don't get the problem. Could you be more precise?

xeonz1, 2021-12-01:
also I got this
[image]

[image]

xeonz1, 2021-12-01:
> I don't get the problem. Could you be more precise?

stuff like this about macros stopped the building

the min in the figure is already defined in minwindef
[image]

jcarpent, 2021-12-01:
You need to include nominmax in the Windows build in CMake.

jcarpent, 2021-12-01:
See https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/src/CMakeLists.txt#L33

xeonz1, 2021-12-01:
> See
> 
> https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/src/CMakeLists.txt#L33

Gracias!