stack-of-tasks/pinocchio #1560: LaTeX formula on the documentation not displayed due to Missing \end{eqnarray} error

traversaro, 2021-12-03:
The math part of the documentation of `pinocchio::forwardDynamics` is not displayed due to an error `Missing \end{eqnarray}`:

[image]

See also https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#ab26c1ca5d01d5885b329fc35793682df .

The related line of code is https://github.com/stack-of-tasks/pinocchio/blob/a1df23c2f183d84febdc2099e5fbfdbd1fc8018b/src/algorithm/contact-dynamics.hpp#L17, but I could not understand at the first sight the problem.

The error is present on both the latest Google Chrome and Firefox. Upon more inspection, it seems to also affect `pinocchio::impulseDynamics`.

jcarpent, 2021-12-03:
Thanks @traversaro for raising this point, pretty useful as usual. I've also noticed this issue on my side without taking time to analyse it.

jcarpent, 2021-12-06:
Solved by #1567.

closed 2021-12-06 by jcarpent