stack-of-tasks/pinocchio #1571: Undefined symbol error in libpinocchio_pywrap.so

spykspeigel, 2021-12-07:
Hello,
I tried a source installation. 
But received the following error
```
  Linking CXX executable test-cpp-serialization
/usr/bin/c++   -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG  -rdynamic CMakeFiles/test-cpp-serialization.dir/serialization.cpp.o  -o test-cpp-serialization  -Wl,-rpath,/home/sp/Documents/pinocchio/build/src ../src/libpinocchio.so.2.6.4 -lboost_unit_test_framework -lboost_filesystem -lboost_system -lboost_serialization -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 
/usr/bin/ld: CMakeFiles/test-cpp-serialization.dir/serialization.cpp.o: in function `__static_initialization_and_destruction_0(int, int) [clone .constprop.0]':
serialization.cpp:(.text.startup+0x1c6): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::basic_cstring<char const>, boost::unit_test::basic_cstring<char const>, unsigned long, boost::unit_test::decorator::collector&)'
/usr/bin/ld: serialization.cpp:(.text.startup+0x252): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
/usr/bin/ld: serialization.cpp:(.text.startup+0x2fa): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
/usr/bin/ld: serialization.cpp:(.text.startup+0x39a): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
/usr/bin/ld: serialization.cpp:(.text.startup+0x43a): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
/usr/bin/ld: serialization.cpp:(.text.startup+0x4da): undefined reference to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)'
/usr/bin/ld: CMakeFiles/test-cpp-serialization.dir/serialization.cpp.o:serialization.cpp:(.text.startup+0x57a): more undefined references to `boost::unit_test::ut_detail::auto_test_unit_registrar::auto_test_unit_registrar(boost::unit_test::test_case*, boost::unit_test::decorator::collector&, unsigned long)' follow
collect2: error: ld returned 1 exit status
make[2]: *** [unittest/CMakeFiles/test-cpp-serialization.dir/build.make:97: unittest/test-cpp-serialization] Error 1
```

On turning off Build_unit_test, the compilation was complete. But on importing Pinocchio I receive the following import error.
```
ImportError: /usr/local/lib/libpinocchio.so.2.6.4: undefined symbol: _ZN5boost6system15system_categoryEv
```
 The dependencies with, 
`ldd /usr/local/lib/libpinocchio.so.2.6.4 ` are

```
	linux-vdso.so.1 (0x00007fffaa9b3000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f2ceca80000)
	liburdfdom_model.so.1.0 => /lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007f2ceca5b000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2cec879000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2cec85e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2cec66c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2cec647000)
	libtinyxml.so.2.6.2 => /lib/x86_64-linux-gnu/libtinyxml.so.2.6.2 (0x00007f2cec62f000)
	libconsole_bridge.so.0.4 => /lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f2cec629000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2cec4da000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2cecafe000)
```

I suspect there is an issue with the libboost but not sure about the specific issue.
Any help is greatly appreciated.

spykspeigel, 2021-12-07:
Hey @jcarpent. Yes, I had a binary installation. But working on a new stateclass in Crocoddyl library and for this I wanted  python binding of dIntegrateTransport. There is no binding related to dIntegrateTransport in Pinocchio. Is it not exposed for any specific reason?
For this requirement, I did add the corresponding bindings but faced the above issue.

jcarpent, 2021-12-07:
So just remove it and let's play.
The `dIntegrateTransport` will be added soon. We already have a fix for that thanks to @ManifoldFR.

jcarpent, 2021-12-07:
I think we can now close the issue which seems to be solved.

closed 2021-12-07 by jcarpent

cmastalli, 2021-12-07:
> The `dIntegrateTransport` will be added soon. We already have a fix for that thanks to @ManifoldFR.

This is great!
These bindings are blocking us, we are keen to implement (or support you) ourselves if this will take time to be available.
Could you tell us when you think it will be available (in devel branch)?

Many thanks ;)

proyan, 2021-12-07:
The python bindings of `dIntegrateTransport` are already present in the main lib: https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/multibody/liegroups.hpp#L194

Is there something specific that you are searching @spykspeigel @cmastalli ?

cmastalli, 2021-12-07:
> The python bindings of `dIntegrateTransport` are already present in the main lib: https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/multibody/liegroups.hpp#L194
> 
> Is there something specific that you are searching @spykspeigel @cmastalli ?

We need the bindings of the joint algo: https://github.com/stack-of-tasks/pinocchio/blob/a1df23c2f183d84febdc2099e5fbfdbd1fc8018b/src/algorithm/joint-configuration.hxx#L174-L220, which I think it should be implemented here: https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/algorithm/expose-joints.cpp

ManifoldFR, 2021-12-07:
Hi, I cherry-picked a commit from my working branch and proposed PR #1572 to add this functionality.