stack-of-tasks/pinocchio #1574: c++ compilation error

PangDaCAT, 2021-12-09:
According to the build from source in Installation Procedure,I installed pinocchio 2.6.4 under ubuntu 18.04. But when I run the example,an error occurred. 
In the function ‘pinocchio::ModelTpl<
double, 0, pinocchio::JointCollectionDefaultTpl>& pinocchio::urdf::buildModel<double, 0, pinocchio::JointCollectionDefaultTpl>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator <char>> const&, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, bool)':
/usr/local/include/pinocchio/parsers/urdf/model.hxx:394: for'pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>> const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)' undefined reference

PangDaCAT, 2021-12-09:
#include <iostream>

#include "pinocchio/parsers/urdf.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
#include "pinocchio/algorithm/jacobian.hpp"
#include "pinocchio/algorithm/rnea.hpp"

// PINOCCHIO_MODEL_DIR is defined by the CMake but you can define your own directory here.
#ifndef PINOCCHIO_MODEL_DIR
  #define PINOCCHIO_MODEL_DIR "/home/tiansk/WenJian/webots/a1_description"
#endif

int main(int argc, char **argv)
{
	//std::cout<<"hello world!"<<std::endl;
  // You should change here to set up your own URDF file or just pass it as an argument of this example.
  const std::string urdf_filename = (argc<=1) ? PINOCCHIO_MODEL_DIR + std::string("/urdf/a1.urdf") : argv[1];
  pinocchio::Model model;
  pinocchio::urdf::buildModel(urdf_filename,model);
}

jcarpent, 2021-12-09:
you need to add the C++11 flags when compiling your file: `g++ -O3 -std=c++11 $(pkg-config --cflags --libs pinocchio) test.cpp -o test` if you file is name `test.cpp`

closed 2021-12-09 by jcarpent

PangDaCAT, 2021-12-09:
This doesn't seem to work.
try2.cpp:(.text.startup+0x68): to'pinocchio::urdf::details::parseRootTree(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char >> const&, pinocchio::urdf::details::UrdfVisitorBaseTpl<double, 0>&)' undefined reference

jcarpent, 2021-12-09:
The reverse then. You need to use std=c++98

PangDaCAT, 2021-12-09:
sorry, the same error..

jcarpent, 2021-12-09:
Could you check in the history of issues. Similar issues appear with the trouble.

SwonGao, 2022-10-28:
hello, have you fixed the issue?

jcarpent, 2022-10-28:
Yes, there was no issue on Pinocchio's side.
You need to correctly set-up the compilation flags ;)

SwonGao, 2022-10-28:
can you help me with it... i got

jcarpent, 2022-10-28:
Your message is incomplete @SwonGao

GoldenSeaC, 2023-10-17:
I resolved this problem(not sure totally) by config the CMakeLists file. 
After installing the pinocchio lib on PC and add the pkg conig path to the environment variable (.bashrc file)
,I apended the following lines to my cmakelists file
------------------------------------------------------------------------------------
find_package(PkgConfig REQUIRED)
pkg_check_modules(PKG_PIN_CONFIG REQUIRED pinocchio)#PKG_PIN_CONFIG is customized name
include_directories(${PKG_PIN_CONFIG_INCLUDE_DIRS})
add_executable(overview-urdf overview-urdf.cpp)
target_link_libraries(overview-urdf ${PKG_PIN_CONFIG_LIBRARIES})
-------------------------------------------------------------------------------------