stack-of-tasks/pinocchio #1575: building error pinocchio3-preview

alme96, 2021-12-13:
Hi all,

I am trying to compile pinocchio but I get always the same error:
[image]
I tried the branches "pinocchio3-preview", "v2.9.1" and "v2.6.4". Is this issue related to pinocchio or to my computer?

Thank you for helping.

proyan, 2021-12-13:
Hi @alme96 
Please include your system information, installation instructions that you followed etc when asking for assistance on an open source repository. If we can't reproduce/understand your setup, no one can help you.

wxmerkt, 2021-12-13:
The above error happens when your system runs out of memory when compiling. It's likely to freeze up just before the compiler gets killed. Try `make -j1` (if you have 16GB and compile with gcc, I would not go higher than `-j2`/`-j3`). The other option would be to use clang which uses less memory during the compilation

jcarpent, 2021-12-13:
Just do a `make -j1` without parallel compilation.

jcarpent, 2021-12-13:
I would also suggest using clang instead of gcc, which is smarter and more efficient

closed 2021-12-13 by jcarpent