stack-of-tasks/pinocchio #1576: how to import pinocchio.casadi

xeonz1, 2021-12-15:
I compiled pinocchio 3 along with casadi support and py bindings. I have 
[image]
But I still cannot use pinocchio.casadi module. What should I do?

jcarpent, 2021-12-15:
Did you install it?
If so, `import pinocchio.casadi as cpin` should work.

xeonz1, 2021-12-15:
> 

It works for me. How should I use this module?

jcarpent, 2021-12-15:
As normal Pinocchio, you can transform a classic model into a casadi one and perform as usual.
```
cmodel = cpin.Model(pin.Model())
cdata  = cmodel.createData()
```

closed 2021-12-15 by jcarpent

xeonz1, 2021-12-15:
> 

What about using symbolic q? I use ConfigVectorType variables and assign SX variables to them in C++. How to do in py?

jcarpent, 2021-12-15:
You can directly pass Casadi Matrix types as usual with Casadi.

xeonz1, 2021-12-18:
> You can directly pass Casadi Matrix types as usual with Casadi.

I use pinocchio.casadi.jointJacobian(robot, data, cs_q, 1) where cs_q is a casadi MX, but it says 

`Python argument types in
    pinocchio.pinocchio_pywrap_default.computeJointJacobian(Model, Data, MX, int)
did not match C++ signature:
    computeJointJacobian(struct pinocchio::ModelTpl<double,0,struct pinocchio::JointCollectionDefaultTpl> model, struct pinocchio::DataTpl<double,0,struct pinocchio::JointCollectionDefaultTpl> {lvalue} data, class Eigen::Matrix<double,-1,1,0,-1,1> q, unsigned __int64 joint_id)
`

jcarpent, 2021-12-18:
It should be a Casadi SX.