stack-of-tasks/pinocchio #1578: Sudden Termination of Python when importing pinocchio.casadi

xeonz1, 2021-12-20:
The Import failed without any message and python just exit. I can still compile c++ codes with pinocchio. What should I do to debug? It still works days ago.

xeonz1, 2021-12-20:
pinocchio can be normally imported. only pinocchio.casadi is unavailable

jcarpent, 2021-12-20:
On which commit are you working now? What was the former one?

xeonz1, 2021-12-20:
> 

I debugged for a while and found that when import pinocchio.casadi a SX is created with value a tuple (0), which may caused exception. I guess it may use import casadi.casadi instead of import casadi

jcarpent, 2021-12-20:
Sorry, but you are hard to follow. 
Could you provide a tiny example that is not working on your side in such a way we can try on our side?

xeonz1, 2021-12-20:
> 

Just import pinocchio.casadi. I see Process finished with exit code -1073741819 (0xC0000005)

jcarpent, 2021-12-20:
I'm using casadi 3.5.5 and everything works perfectly well.
If it was working well a few days ago, it should remain the case. I suspect some change on your side.

Please consider reinstalling a clean version in a clean environment.

closed 2021-12-20 by jcarpent