stack-of-tasks/pinocchio #1579: Using RigidConstraintModelTpl in new pinocchio3-preview branch

manumerous, 2021-12-30:
Hi 

I would like to use the new impulse-dynamics in the pinocchio3-preview branch for the foot impact of a biped and I therefore need to use the new `RigidConstraintModelTpl` and `RigidConstraintDataTpl` in [pinocchio/algorithm/contact-info.hpp](https://github.com/stack-of-tasks/pinocchio/blob/pinocchio3-preview/src/algorithm/contact-info.hpp)

I was a bit confused by the required inputs of the RigidConstraintModelTpl constructor and I am not sure what `joint1` and `joint2` should represent and did not find any documentation on the topic. I have a point contact model and therefore only a single joint in contact. 

Should I therefore use the following constructor?

```  
/// \brief Contructor with from a given type, joint1_id and placement.
///
/// \param[in] type Type of the contact.
/// \param[in] joint1_id Index of the joint 1 in the model tree.
/// \param[in] joint1_placement Placement of the constraint w.r.t the frame of joint1.
/// \param[in] reference_frame Reference frame in which the constraints quantities are expressed.
///
RigidConstraintModelTpl(const ContactType type,
                     const JointIndex joint1_id,
                     const SE3 & joint1_placement,
                     const ReferenceFrame & reference_frame = LOCAL)
```

If that is the case, why is there an option to specify two joints for the `CONTACT_3D` point foot option?

Any help regarding this would be much appreciated. 
Best regards, 
Manuel

proyan, 2021-12-30:
Hi Manuel,
The joint1 and joint2 correspond to the two joints on the kinematic tree which are in contact. For example, if we put the hands together in a "Namaste" sign, that would mean joint 1 corresponds to left hand joint, and joint 2 corresponds to the right hand joint.

The origin corresponds to the joint 0 in the multibody tree. Since you are making a contact with the ground, which is fixed w.r.t the origin, the joint2_id in your case would be 0, and joint1_id would be the ankle/foot joint

manumerous, 2021-12-30:
Hi Rohan, 

I have not thought about the possibility of contacts between two joints of the robot. it makes a lot of sense now. 
Thank you very much for the great response!

closed 2021-12-30 by manumerous