stack-of-tasks/pinocchio #1580: Extremly long build time for pinocchio3-preview

manumerous, 2021-12-31:
Hi, 

While trying out hte new pinocchio3-preview branch I observed that the build time is extremely long. To build the package using `catkin build pinocchio` it takes me 1h and 15 min to build compared to the about 20 seconds (excluding hpp-fcl) it takes me to build the current master branch. I found this stark difference quite strange.

I was wondering if you experienced similar build times? Are you aware of a solution to shorten the build time?

FYI I compiled on a Ubuntu 20.04 LTS system with a AMD Ryzen Pro 7. 

I hope you don't mind my question and any help is appreciated. 

Best, Manuel

jcarpent, 2021-12-31:
I'm very surprised by your 20 seconds. It seems there is no compilation at all as the package is already compiled. You should start from scratch.
Concerning Pinocchio 3, we have added a lot of new features, among them extended Python bindings, constrained dynamics support, etc. with a lot of additional unit testings to validate implemented algorithms.

My first advice would be to deactivate the compilation of unit tests.

jcarpent, 2021-12-31:
By the way, we are adding the template instantiation within the main library to really shorten the whole compilation timings of the project.

manumerous, 2021-12-31:
I achieve the 20s build time when I first clean my catkin workspace using `catkin clean` followed by `catkin build pinocchio`. 
Could it be that the preinstalled `robotpkg-pinocchio` and `robotpkg-py38-pinocchio` apt packages (which i have installed) contain most of the already compiled components for building pinocchio master? 

I will give the deactivation of the unit tests a try. 
Thanks for your help and continued work on this project. It's great! I am really looking forward to see where pinocchio will go in the new year. 

A la nouvelle année!

jcarpent, 2022-01-02:
I do not know how catkin operates, but they might be a catch mechanism behind it.
Catkin can also install the binary version built on the building farm of ROS.

PS: thanks for the kind and french thought.

closed 2022-01-03 by manumerous

wxmerkt, 2022-01-04:
Hi @manumerous 
Happy New Year!

Is it possible that you are using ccache [first hunch] or not correctly clearing the package? It definitely won't pick up any binary installed Pinocchio versions, whether robotpkg or ROS, to reduce buildtime. You can get rid off the ccache cache by removing `~/.ccache` for more reliable compile timings from scratch, or deactivating ccache temporarily.

`catkin build` will build Pinocchio in its default CMake configuration, i.e. with unittests and examples activated and should take quite some time due to that (I'm seeing ~20min with Clang with unittests off I believe). To reduce the time, you can either set the options via CMake args with `catkin config --cmake-args -DBUILD_EXAMPLES=OFF -DBUILD_TESTING=OFF` for all projects or edit the `CMakeLists.txt` `option` settings in Pinocchio. I also recommend setting Clang as the compiler, which can also be done for all packages with `catkin config --cmake-args -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang`.

Best wishes,
Wolfgang

manumerous, 2022-01-12:
Thank you for your comments. They were rally helpfull!