stack-of-tasks/pinocchio #1581: Model.addJoint does not test the input joint Id

florent-lamiraux, 2022-01-03:
I am using `pinocchio` for teaching robotics to students. Building a kinematic chain, one of them made `python` crash by simply adding a joint with a wrong parent joint id. The following code reproduces the bug:
```
import pinocchio

model = pinocchio.Model()
model.addJoint(1, pinocchio.JointModelRX(),pinocchio.SE3.Identity(), 'x')
```

jcarpent, 2022-01-04:
You're totally right @florent-lamiraux. We should put more safe-guard in the Model of Pinocchio, following existing discussions in #1025.

proyan, 2022-01-07:
From #1025, we never really fixed the way forward. The propositions of discussion that I see is:
1) Adding a macro to throw, which is by default set to False (needlessly complex).
2) Similar to (1), throwing with default argument in the function (needlessly complex)
3) Throwing by default (breaks API)
4) Warning messages (pollutes cout)

I would be in favor of throwing by default, but that breaks the behavior in pinocchio3 compared to pinocchio2. Further continuation of discussion in #1025 would be welcome here.

florent-lamiraux, 2022-01-07:
Unlike #1025, calling `addJoint` with a wrong parent joint Id makes the program crash. So I think there is no problem in throwing by default.

jcarpent, 2022-01-07:
I do agree with Florent.

proyan, 2022-01-07:
> Unlike #1025, calling `addJoint` with a wrong parent joint Id makes the program crash. So I think there is no problem in throwing by default.

Makes sense. I can provide a solution to this instead of linking with the previous issue then.

jcarpent, 2022-01-10:
Fixed by #1586.

closed 2022-01-10 by jcarpent