stack-of-tasks/pinocchio #1587: Is GeometryObject::ngeoms really useful ?

florent-lamiraux, 2022-01-10:
https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/src/multibody/geometry.hpp#L168
Is there any difference between `ngeoms` and `geometryObjects` ?
If no, the first one should be deprecated and replaced by a const reference to the second one.
For some robotics demo, I add and remove geometries from my model and I got confused since `geometryObjects.erase(it)` needs to be coupled with `ngeoms--`.

By the way, I wrote a function that removes a `GeometryObject` instance from a `GeometryModel` and remove the corresponding collision pairs. The input of the function is the name of the object. Do you think this function should be added to pinocchio ?

jcarpent, 2022-01-10:
> By the way, I wrote a function that removes a `GeometryObject` instance from a `GeometryModel` and removes the corresponding collision pairs. The input of the function is the name of the object. Do you think this function should be added to pinocchio ?

Yes, of course.

jcarpent, 2022-01-10:
> Is there any difference between `ngeoms` and `geometryObjects` ?
> If no, the first one should be deprecated and replaced by a const reference to the second one.
> For some robotics demo, I add and [...]

Why not. How do you see the solution?

florent-lamiraux, 2022-01-10:
In class `GeometryModel`:
```
const Index& ngeoms;
```
and in the constructor:
```
    GeometryModel()
    : ngeoms(geometryObjects.size())
    , geometryObjects()
    , collisionPairs()
    {}
```
If the compiler accepts, this should do the job.

florent-lamiraux, 2022-01-10:
> It will lead to some memory leaks I guess, as `std::vector<T>::size` follows this signature:
> 
I do not think so. There is no memory allocation involved.

jcarpent, 2022-01-10:
Yes, but the return object is a temporary variable: it is not held by the std::vector object.

florent-lamiraux, 2022-01-10:
Let see what the compiler thinks about it. I will try tomorrow.

jcarpent, 2022-01-10:
Let's then bet a 🍺  dear Florent!?

florent-lamiraux, 2022-01-11:
I think you are right. My solution won't work. Then I will propose a PR to add a method that removes an object. Calling this method, users will not need to handle variable `ngeoms`.

closed 2022-02-15 by jcarpent