stack-of-tasks/pinocchio #1591: Remove Body

JonathanKuelz, 2022-01-20:
Hey,
I am working on modular robotics with pinocchio, which means I build lots of robots composed of the same modules. It would be convenient if it was possible to add and remove bodies and joints for this purpose (as mentioned in #1451 ) instead of building every combination from scratch.

I am only working with the python wrappers, where I was able to add as many joints / bodies as I want. However, removing parts of the kinematic chain fails as I have no access to the n-{bodies, joints, frames, qs, vs} properties of a model. Is it somehow possible to decrement these read-only properties in python? I believe I have the rest of the removing procedure figured out, so this would be a workaround until removing parts from the kinematic chain will be properly supported in pinocchio.

jcarpent, 2022-01-21:
Indeed, it would be a nice feature to add to Pinocchio
The only missing part is the need of rebuilding a Data struct as soon as the model has changed. Do you have in mind typical applications for that?

JonathanKuelz, 2022-01-21:
My personal application for that is the evaluation of different modular robots for a given task - like e.g. finding a good combination from a large set of possible modules to reach multiple poses in a work space. ([Whiman et al.](https://www.semanticscholar.org/paper/Modular-Robot-Design-Synthesis-with-Deep-Learning-Whitman-Bhirangi/290780f81eb42ba0d22f5335e7eed53b3829e0c8) have done something like it for example, but they only expanded their robot while I would like to change it)

Or suppose you want to calculate the forward kinematics for a set of different grippers you are thinking about attaching to your robot - it would be more intuitive to have `robot.removeLast(), robot.addNew(gripper)` functionalities than to build a new robot for every gripper from scratch.

Whenever you have multiple alternatives for how to build your robot - be it a humanoid for which you want to find the ideal morphology or a modular industrial robot - I think being able to change the robot instance would be of benefit.