stack-of-tasks/pinocchio #1593: Problem about joint jacobian reference frame and floating base pose

xeonz1, 2022-01-22:
I have a floating base model with root_joint=pinocchio.JointModelFreeFlyer()
1. In my practice, oMf.np of floating_base frame is an identity matrix, while I think it should be up to global_quaternion and global position. Why is universe aligned with floating base? I am a little confused about the gravity direction. Is it still in z-axis downward direction, i.e, [0,0, -9.81]?
2. If root_joint connects the WORLD and the floating base, why is J(root_joint, LOCAL)v different from getFrameVelocity(root_joint, LOCAL)? Aren't they expressed in the same frame? If not the same frame, what are the two LOCAL frames?
Seems like the connection in pinocchio is a bit different from the URDF definition.

jcarpent, 2022-01-23:
You have provided an incomplete example. We don’t have access to the model.
@proyan could you provide help on this issue please?

xeonz1, 2022-01-23:
> You have provided an incomplete example. We don’t have access to the model. @proyan could you provide help on this issue please?

Sorry, I uploaded the urdf in: 
https://github.com/xeonz1/hello-world/blob/master/robot.urdf

xeonz1, 2022-01-24:
I may have solved the problem. Analysis is given by:
1. The generated quaternion may be irregular and lead to wrong SE3 transform. Therefore one can use:
```
euler2quat([random.random() for i in range(3)])
```
2. One should use only the joint velocity v_j instead of the generalized velocity of the model.

closed 2022-01-24 by xeonz1

proyan, 2022-01-24:
Glad you solved the problem. Regarding your analysis, 
1a) there is a `randomConfiguration(model, lowerLimits, upperLimits)` function available in pinocchio that you could use to generate random configurations, without worrying about quaternion normalization
1b) you could also use `.normalize()` function in quaternion in order to normalize the randomly generated quaternion.

Have fun using pinocchio :)