stack-of-tasks/pinocchio #1596: Problem about JointModelFreeFlyer initial placement

xeonz1, 2022-01-26:
Hi,

I want to ask if the initial placement of a JointModelFreeFlyer is related to the definition of floatingbase in urdf? For example:
```
<link
    name="floating_base">
    <inertial>
      <origin
        xyz="0.00024495 -0.0012722 -0.067949"
        rpy="0 0 0" />
```

jcarpent, 2022-01-26:
Sorry, but your issue is not clear at all.
I don't get what do you expect.

xeonz1, 2022-01-26:
> Hi @xeonz1 The `origin` in the `link` corresponds to the position of the `link` w.r.t it's parent. So for a zero joint configuration, the `floating_base` should appear at the position specified in the `origin` in your example

Thanks for the reply. I have a floating_base link defined as 
```
    <inertial>
      <origin
        xyz="0.00024495 -0.0012722 -0.067949"
        rpy="0 0 0" />
```
Then I run
```
import pinocchio
import numpy as np
from quat2euler import *
robot, robot_collison, robot_visual = pinocchio.buildModelsFromUrdf(
    "./robotfb.urdf", root_joint=pinocchio.JointModelFreeFlyer()
)
data = robot.createData()
q = np.array([0, 0, 0]+ euler2quat_math([0]*3))
print(pinocchio.centerOfMass(robot, data, q))
```
with the output 
```
[0.00024495 0.0012722  0.067949  ]
```
Why it is not the same as <inertial><origin xyz> and appears that the floating_base is revoluted by 180 degree w.r.t. its x-axis?

jcarpent, 2022-01-26:
It depends on your convention for Euler angles I would say.
Use the eigenpy (Eigen Python bindings) to get the correct convention.
You can also use `pin.SE3ToXYZQuat(pin.SE3.Identity())`

xeonz1, 2022-01-26:
> It depends on your convention for Euler angles I would say. Use the eigenpy (Eigen Python bindings) to get the correct convention. You can also use `pin.SE3ToXYZQuat(pin.SE3.Identity())`

For 
```
print(pinocchio.se3ToXYZQUAT(pinocchio.SE3().Identity()))
```
I get [0. 0. 0. 0. 0. 0. 1.]
and for my `euler2quat_math` I get for non-rotation quaternion [1,0,0,0]
What is the definition of quaternion in pinocchio?

jcarpent, 2022-01-26:
This is the one of Eigen (x,y,z,w).

xeonz1, 2022-01-26:
> This is the one of Eigen (w,x,y,z).

Ok. My problem is solved.

closed 2022-01-26 by xeonz1