stack-of-tasks/pinocchio #1597: model.nbodies != number of body frames

JonathanKuelz, 2022-01-31:
Hello,
this is issue is partly issue, but also a question:

while playing around with runtime modification of pinocchio models in Python (see #1591 ), I realized that the model parameter nbodies is not equal to the number of body frames. At first I thought this could be explained by multiple frames being attached to a single body (could imagine reasons for that). However, seeing that the methods `existBodyName` and `getBodyId` work on a frame basis, I am confused about what exactly a body in pinocchio is.

As example, here are the body frame names from a model representing a Universal Robot (UR5) I loaded from URDF:
['world', 'base_link', 'base', 'shoulder_link', 'upper_arm_link', 'forearm_link', 'wrist_1_link', 'wrist_2_link', 'wrist_3_link', 'ee_link', 'tool0']

11 names, model.existBodyName(N) returns true for every name N in this list. (model.getBodyID() works for all of them aswell)

However, `model.nbodies = 7` (number of inertias: also 7)

As far as I understand it, there are two "representations" of bodies in a robot model: The "real bodies", which are added using appendBodyToJoint() and then used for dynamic calculations - and the "body frames". However, for me the separation is not really clear. It would be kind if you could help me by either telling me how my described understanding is wrong, or giving me a hint how to deal with the following questions:
- Should nbodies always equal njoints? If not, how can I efficiently see which of my 7 (UR5 example) bodies is connected to which joint. (I can do that easily for the frames, but actually: Which frames would belong to which body if model.njoints != model.nbodies?)
- If nbodies == njoints, what would be the right way to add a body (e.g. payload) to an already existing model? Just changing the according inertia? (Using the addBodyToJoint method would increase the model's body count)
- Having 7 inertias and 11 body frames (above): Can I know which body (body as in "body frame") contributed to which inertia?

jcarpent, 2022-01-31:
Bodies are indeed physical bodies defined by spatial inertia.
Both `existBodyName` and `getBodyId`operates from Frame with BODY type.

For real applications, you don't really need the notion of `nbodies` which is a counter incremented when calling `addBodyToJoint`.
In Pinocchio, we cumulate the body inertia to the inertia supported by a given joint.

In addition in Pinocchio, if you append a BODY frame (via `addFrame`) with an inertia, then this latest will be added to the joint inertia supporting the given frame.

Hope it this answer will be help you.

JonathanKuelz, 2022-01-31:
Thanks, so could wrap it up as follows: `nbodies` is a helper that's not really necessary, inertias are updated automatically when adding a frame and "one body, one frame" more or less sums it up.

Assuming that's right, I'm gonna close this issue as this allows me to continue working with bodies in the model 👍🏾

closed 2022-01-31 by JonathanKuelz