stack-of-tasks/pinocchio #1598: Wrong index in ABAChecker

kozakromch, 2022-01-31:
ABAChecker Has the following code
```
    bool checkModel_impl(const pinocchio::Model& model) const {
        {
            typedef typename pinocchio::Model::JointIndex JointIndex;

            for (JointIndex j = 1; j < (JointIndex)model.njoints; j++)
                if ((model.inertias[j].mass() < 1e-5) ||
                    (model.inertias[j].inertia().data()[0] < 1e-5) ||
                    (model.inertias[j].inertia().data()[3] < 1e-5) ||
                    (model.inertias[j].inertia().data()[5] < 1e-5)) return false;
```

I think there is an index error and instead `
(model.inertias[j].inertia().data()[3] < 1e-5) ||`
should be `(model.inertias[j].inertia().data()[2] < 1e-5) ||`

As my checker falling at my model with following debugging data 
 joint = 3  m = 9.475
  c = -2.92879e-06 -1.28519e-06      14.5843
  I = 
     1639.16 -2.00142e-13  7.67946e-06
-2.00142e-13      1639.16 -7.68183e-06
 7.67946e-06 -7.68183e-06      1934.75
 Data 
     1639.16
-2.00142e-13
     1639.16
 7.67946e-06
-7.68183e-06
     1934.75
model.inertias[j].mass() < 1e-5) = 0
(model.inertias[j].inertia().data()[0] < 1e-5) = 0
(model.inertias[j].inertia().data()[3] < 1e-5) = 1
(model.inertias[j].inertia().data()[5] < 1e-5) = 0

jcarpent, 2022-01-31:
You're totally right. I fixed it here #1599.

closed 2022-01-31 by kozakromch

jcarpent, 2022-01-31:
@kozakromch Thanks a lot for raising this issue.
Do you run Pinocchio in debug mode then?

kozakromch, 2022-01-31:
@jcarpent Yes, I run pinocchio in Debug mode.
But i catch this error at checking my model