stack-of-tasks/pinocchio #1600: Jacobian for q1 \ominus q0

stevenjj, 2022-02-01:
Hi,
Perhaps this is straightforward, but I'm unable to figure this out. I have two configuration vectors, q0 and q1. I know that I can find the tangent vector that brings q0 to q1 using the difference function, namely:

difference(q0, q1) = v, such that  q1 = integrate(q0, v)

However, what I'm trying to achieve is a Jacobian for the difference function so that
v = J*(q1-q0) where J is fixed about q0 and varies with q1 and that J should be size (model.nv x model.nq). Can this be constructed with any of the existing functions? I thought that dIntegrate is something that I could use, but it outputs a matrix of size (model.nv x model.nv) and requires \Delta q to be in tangent space already. The reason this is tricky is that model.nq > model.nv due do the presence of spherical joints represented as quaternions.

I do plan on moving on towards working with tangent spaces, but I'm just wondering if this exists in some capacity before I move on.
Thank you!

proyan, 2022-02-01:
Hi Steven

There is also a dDifference that seems to be what you are looking for

Best

jmirabel, 2022-02-01:
You seem confused about some facts.
- In `v = J*(q1-q0)`, `J` has to have `model.nv` columns.
- The Jacobian `J` of the difference `q1 - q0` wrt `q1` is such that `[ (q1 + v) - q0 ] - [ q1 - q0 ]` is equivalent to `J v` when the norm of `v` tends to zero.

stevenjj, 2022-02-01:
Thanks for getting back to me @proyan  @jmirabel! 
@proyan I thought so too, but dDifference also operates in tangent spaces only for Delta q.

@jmirabel I don't think I I follow.  Let's take the case of a single spherical joint which would be represented as a quaternion. 
Here q is size 4, and v would be size 3. What I'm trying to achieve is a linear approximation of v = q1 \ominus q0 w.r.t. q0.  So J in v = J*(q1-q0) needs to be 3x4. I'm not performing tangent space difference, I'm simply doing a subtraction operation. Perhaps my approach silly and I may just reformulate the rest of my work to be in tangent space.

jmirabel, 2022-02-01:
I understand what you want. This isn't implemented in Pinocchio. If you want to stick to the coeff wise formulation, I advise you to write a proper limited development. `dv = J1 dq1 + J2 dq2` where `dq` are understood coeff wise.

Now, if you go down this road, you are on your own. Pinocchio will not be of much help.

stevenjj, 2022-02-01:
OK. Thank you very much for the clarification. I just wanted to make sure that the function didn't exist.

closed 2022-02-01 by stevenjj

nmansard, 2022-02-02:
In case it might be useful, there was a class about this question, 
material and video available here:

https://memory-of-motion.github.io/summer-school/materials