stack-of-tasks/pinocchio #1601: Scalar of Type Float for Code Generation

plancherb1, 2022-02-02:
Hi, I've been trying to use the code generation features and have found that for some reason I can only get them to support a ```Scalar``` of type ```double```. Is there any way to get them to accept a ```Scalar``` of type ```float```? 

For example setting the type parameter ```T``` in the below code snippet to ```double``` works fine, but setting it to ```float``` results in a compilation error stating that there is:
```no matching constructor for initialization of 'CodeGenRNEA<float>```
which is I think caused by this other error:
```note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'pinocchio::Model' (aka 'ModelTpl<double>') to 'const pinocchio::CodeGenRNEA<float>' for 1st argument  struct CodeGenRNEA : public CodeGenBase<_Scalar>```.

Thanks in advance, I assume I am missing something simple!

```
// Import URDF model and set gravity
Model model;
pinocchio::urdf::buildModel(urdf_filepath,model);
model.gravity.linear(Eigen::Vector3d(0,0,-9.81));

// generate the code_gen
CodeGenRNEA<T> rnea_code_gen(model);
rnea_code_gen.initLib();
rnea_code_gen.loadLib();
```

jcarpent, 2022-02-02:
Hi @plancherb1!
You just need to cast your input model as follows and similarly to Eigen:  
```
CodeGenRNEA<T> rnea_code_gen(model.cast<T>());
```

jcarpent, 2022-02-02:
I've closed the issue as the issue seems to be solved. Feel free to reopen it if needed

closed 2022-02-02 by jcarpent