stack-of-tasks/pinocchio #1605: `BaseVisualizer` shouldn't have its own data ?

vnghia, 2022-02-05:
Hi everyone !

I find it unexpected that the method `rebuildData` creates a new copy of the data:
https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/bindings/python/pinocchio/visualize/base_visualizer.py#L40-L43

Inside `RobotWrapper`, if I call `rebuildData` on it, I will get 2 source of data, one inside that instance and one inside its `viz`.
https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/bindings/python/pinocchio/robot_wrapper.py#L234-L239

I think we should have only one source or am I missing something ? I could send a pull request if you want, probably just something simple like:
```python
     def rebuildData(self): 
         """Re-build the data objects. Needed if the models were modified. 
         Warning: this will delete any information stored in all data objects.""" 
         self.data, self.collision_data, self.visual_data = createDatas(self.model, self.collision_model, self.visual_model) 
         if self.viz is not None: 
             self.viz.data = self.data
             self.viz.collision_data = self.collision_data
             self.viz.visual_data = self.visual_data 
```

jcarpent, 2022-02-05:
Thanks for raising this question @vnghia.
I would say that your solution should also check that initially, the data between the viewer and the RobotWrapper were shared or not.
Could you add this feature by checking the addresses of the objects?

vnghia, 2022-02-05:
@jcarpent It is actually a deeper problem. Normally, the user will call this function:
https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/bindings/python/pinocchio/robot_wrapper.py#L265-L272
which leads to
https://github.com/stack-of-tasks/pinocchio/blob/593d4d43fded997bb9aa2421f4e55294dbd233c4/bindings/python/pinocchio/visualize/base_visualizer.py#L12
Unfortunately, not all parameters of `__init__` are passable. So by default, the models will always be shared and the data won't be. We could do something like:
```python
GepettoVisualizer(self.model, self.collision_model, self.visual_model, kwargs.pop("data", None) #...)
```
What do you think about this ?

jcarpent, 2022-02-05:
You can force them to be equal in the initViewer by adding an argument shared_data. Is it clear?

jcarpent, 2022-02-07:
Solved by #1606.

PR #1606 (2022-02-07): allow `robot_wrapper` to share data with its `viz`
For your information, @jcarpent 
Close #1605
closed 2022-02-07 by jcarpent