stack-of-tasks/pinocchio #1608: JointTorqueRegressor

JunningHuang, 2022-02-14:
Hi, team of pinocchio, thanks for the powerful library first!

Now I am trying to do some SysID to identify the inertia parameters, and I noticed that there is a function called "computeJointTorqueRegressor" which utilize the linearity between torque and dynamic parameters (without the friction term). I think the computation of the regression matrix is referred to chapter 6.3 Inertia parameter estimation in the Springer handbook of robotics by Bruno Siciliano. 

Here I have two questions regarding the identification. 
First, since only the torque about the joint rotation axis can usually be measured, do you provide any API to retrieve the joint rotation axis? 
Second, when computing the regression matrix, usually you should compute the transformation matrix from joint 1 to joint 2, joint 1 to joint 3 etc, but it seems the current interface provides only the transformation matrix that relative to the parent frame or to the world frame. Is there any API that I miss in the documentation can retrieve the transformation matrix from joint i to any joint j? or I can just use the intermediate results and compute it?

jcarpent, 2022-02-15:
@JunningHuang Thanks for your interest in Pinocchio.
From your questions, it is not clear why do you need the joint rotation axis or the relative transformation between two joints. Could you depict your need?

As you mentioned, the joint torque regressor maps the dynamic parameters of the links to the joint torque following the relation: https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a144858691dd4b67aa44d01f905088344

JunningHuang, 2022-02-16:
Thanks for the quick reply and the link of the documentation. The document is well explained for the regressor.

Now I am trying to implement the regressor myself, I follow the equations in chapter 6.3 Inertia parameter estimation in the Springer handbook of robotics by Bruno Siciliano, page 122-124. It would require the joint rotation axis and the relative transformation between two joints. 

But I guess the details are in your implementation, so the best way might be read your code.

JunningHuang, 2022-02-16:
> @JunningHuang Thanks for your interest in Pinocchio. From your questions, it is not clear why do you need the joint rotation axis or the relative transformation between two joints. Could you depict [...]

JunningHuang, 2022-02-16:
By the way, have you considered add the damping term in the dynamic parameters? And the regressor would need an extra column for joint velocities.

jcarpent, 2022-02-16:
This is a good point. We should add it to the estimation stage indeed.

JunningHuang, 2022-02-16:
Another question is, I loaded an urdf file of a furuta pendulum to test the interface of dynamic parameters, namely toDynamicParameters(). But it seems there are some terms in the vectorized inertia not compatible with the value retrieve by model.inertias. I guess the terms in the vectorized inertia should be exactly the same with the model.inertias except the inertia of the universe joint.

closed 2022-02-21 by JunningHuang