stack-of-tasks/pinocchio #1611: Casadi python issue

ddliu365, 2022-02-26:
Hi, folks,

Here I report two issues for the Casadi Pinocchio Python module.

I found an issue with the `rpy` class for the Casadi Pinocchio Python interface.

`
from pinocchio.casadi.rpy import *
rpyToMatrix(cs.SX.zeros(3))

`

the log is:

# Python argument types in
    pinocchio.pinocchio_pywrap_casadi.rpy.rpyToMatrix(SX)
did not match C++ signature:
    rpyToMatrix(Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> > rpy)
    rpyToMatrix(double roll, double pitch, double yaw)

Another issue occurs when I tried to access the frame position like previous Pinocchio
`
cs_q = casadi.SX.sym("cs_q", robot.nq)
cs_v = casadi.SX.sym("cs_v", robot.nv)
robot = pin_robot.cmodel
data = pin_robot.cdata

pinocchio.casadi.forwardKinematics(robot, data, cs_q, cs_v)
pinocchio.casadi.computeJointJacobians(robot, data, cs_q)
pinocchio.casadi.updateFramePlacements(robot, data)

data.oMf[robot.getFrameId("l_foot")]
`
after running the program in the jupyter notebook, the server shut down immediately.

jcarpent, 2022-02-26:
Thanks for the report. The first issue was known and fixed in the future Pinocchio 3 which should come soon.
For the second one, please provide a complete test.

ddliu365, 2022-02-26:
After rebooting the machine, I found the second error disappears. Look forward to the coming update.

closed 2022-02-26 by ddliu365