stack-of-tasks/pinocchio #1615: program compilation problem with ROS

skywoodsz, 2022-03-15:
Hi, I want to use pinocchio in ROS. However, when I  compile the program with catkin_make, I get the following error:

In file included from /opt/openrobots/include/pinocchio/fwd.hpp:25:0,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:9,
                 from /opt/openrobots/include/pinocchio/multibody/model.hpp:9,
                 from /opt/openrobots/include/pinocchio/parsers/urdf.hpp:9,
                 from /home/skywoodsz/MQuadruped_ws/src/champ_exter_control/src/pinocchio_control_test.cpp:6:
/opt/openrobots/include/pinocchio/container/boost-container-limits.hpp:30:9: error: #error "You should include pinocchio before the Boost headers (e.g. #include <pinocchio/fwd.hpp>)"
       # error "You should include pinocchio before the Boost headers (e.g. #include <pinocchio/fwd.hpp>)"

I can compile the program with make. My system version is ubuntu1604, ROS-Kinetic and Pinocchio2.6.5. Could you tell me how to fix this problem? Thanks!

nim65s, 2022-03-15:
1. Upgrade your system. 16.04/kinetic is way too old for anything.
2. Don't mix robotpkg binaries and ROS stuff, as pinocchio and its dependencies are provided by both and they are usually not compatible with each other.
3. Read the error message. It says that you should check in `/home/skywoodsz/MQuadruped_ws/src/champ_exter_control/src/pinocchio_control_test.cpp` that the first header you include is `#include <pinocchio/fwd.hpp>`, especially before anything from boost or anything that could include boost.

wxmerkt, 2022-03-15:
To add to (3), include `#include <pinocchio/fwd.hpp>` _prior_ to the `ros.h` header since ROS implicitly includes Boost. Here is a minimal example for how to use Pinocchio with ROS: https://github.com/wxmerkt/pinocchio_ros_example