stack-of-tasks/pinocchio #1619: Joint limit

bhtxy0525, 2022-03-21:
Hey,
I wonder if joint limits in _.urdf file_ have any effect on the inverse kinematics solution？
I noticed that if no limits were added in _.urdf file_, the joint angles obtained by CLIK would exceed 2π

jcarpent, 2022-03-21:
What is CLIK?
You should use a dedicated QP solver to properly account for the joints limits for IK.

closed 2022-03-21 by jcarpent

bhtxy0525, 2022-03-21:
@jcarpent  the example code of inverse kinematics in pinocchio document is called CLIK.
So, the joint limits in _.urdf file_  do not affect the final inverse solution? I want to confirm this.
Looking forward to your reply！
Thank you very much.

jcarpent, 2022-03-21:
Yes.

bhtxy0525, 2022-03-21:
@jcarpent Thank you.
I will consider the QP-based solver.