stack-of-tasks/pinocchio #1620: Broken collision support with latest changes in hpp-fcl

cmastalli, 2022-03-21:
Dear all,

I just updated `hpp-fcl` with the latest devel branch.
After that, I encounter the following compilation issue in Pinocchio:

```bash
[  4%] Building CXX object src/CMakeFiles/pinocchio.dir/parsers/urdf/geometry.cpp.o
[  4%] Building CXX object src/CMakeFiles/pinocchio.dir/parsers/urdf/model.cpp.o
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/deprecation.py
cd /home/cmastalli/devel/pinocchio/build/tmp/src && /usr/bin/clang++-10  -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -Dpinocchio_EXPORTS -I/home/cmastalli/devel/pinocchio/build/tmp -I/home/cmastalli/devel/pinocchio/build/tmp/include -I/home/cmastalli/devel/pinocchio/include -isystem /usr/include/eigen3 -isystem /home/cmastalli/install/include -isystem /opt/ros/melodic/include  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -fPIC   -std=gnu++11 -o CMakeFiles/pinocchio.dir/parsers/urdf/model.cpp.o -c /home/cmastalli/devel/pinocchio/src/parsers/urdf/model.cpp
cd /home/cmastalli/devel/pinocchio/build/tmp/src && /usr/bin/clang++-10  -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DHPP_FCL_HAS_OCTOMAP -DHPP_FCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -Dpinocchio_EXPORTS -I/home/cmastalli/devel/pinocchio/build/tmp -I/home/cmastalli/devel/pinocchio/build/tmp/include -I/home/cmastalli/devel/pinocchio/include -isystem /usr/include/eigen3 -isystem /home/cmastalli/install/include -isystem /opt/ros/melodic/include  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -fPIC   -std=gnu++11 -o CMakeFiles/pinocchio.dir/parsers/urdf/geometry.cpp.o -c /home/cmastalli/devel/pinocchio/src/parsers/urdf/geometry.cpp
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/utils.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/robot_wrapper.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/romeo_wrapper.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/explog.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/shortcuts.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/visualize/__init__.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/visualize/base_visualizer.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/visualize/gepetto_visualizer.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/visualize/meshcat_visualizer.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/visualize/panda3d_visualizer.py
cd /home/cmastalli/devel/pinocchio/build/tmp/bindings/python && /usr/bin/python /home/cmastalli/devel/pinocchio/cmake/compile.py /home/cmastalli/devel/pinocchio/bindings/python /home/cmastalli/devel/pinocchio/build/tmp/bindings/python pinocchio/visualize/rviz_visualizer.py
make[2]: Leaving directory '/home/cmastalli/devel/pinocchio/build/tmp'
[  4%] Built target compile_pyc__home_cmastalli_devel_pinocchio_bindings_python
/home/cmastalli/devel/pinocchio/src/parsers/urdf/geometry.cpp:168:22: error: no viable overloaded '='
            geometry = bvh->convex;
            ~~~~~~~~ ^ ~~~~~~~~~~~
/usr/include/boost/smart_ptr/shared_ptr.hpp:547:18: note: candidate function not viable: no known conversion from 'shared_ptr<hpp::fcl::ConvexBase>' to 'const boost::shared_ptr<hpp::fcl::CollisionGeometry>' for 1st argument
    shared_ptr & operator=( shared_ptr const & r ) BOOST_SP_NOEXCEPT
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:556:18: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::shared_ptr'
    shared_ptr & operator=(shared_ptr<Y> const & r) BOOST_SP_NOEXCEPT
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:567:18: note: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::auto_ptr<Y> & r )
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:576:18: note: candidate template ignored: could not match 'auto_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::auto_ptr<Y> && r )
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:598:18: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr & operator=( std::unique_ptr<Y, D> && r )
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:607:18: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr & operator=( boost::movelib::unique_ptr<Y, D> r )
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:658:18: note: candidate function not viable: no known conversion from 'shared_ptr<hpp::fcl::ConvexBase>' to 'boost::shared_ptr<hpp::fcl::CollisionGeometry>' for 1st argument
    shared_ptr & operator=( shared_ptr && r ) BOOST_SP_NOEXCEPT
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:665:18: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::shared_ptr'
    shared_ptr & operator=( shared_ptr<Y> && r ) BOOST_SP_NOEXCEPT
                 ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:683:18: note: candidate function not viable: no known conversion from 'shared_ptr<hpp::fcl::ConvexBase>' to 'boost::detail::sp_nullptr_t' (aka 'nullptr_t') for 1st argument
    shared_ptr & operator=( boost::detail::sp_nullptr_t ) BOOST_SP_NOEXCEPT
                 ^
/home/cmastalli/devel/pinocchio/src/parsers/urdf/geometry.cpp:170:22: error: no viable overloaded '='
            geometry = bvh;
            ~~~~~~~~ ^ ~~~
/usr/include/boost/smart_ptr/shared_ptr.hpp:547:18: note: candidate function not viable: no known conversion from 'hpp::fcl::BVHModelPtr_t' (aka 'shared_ptr<hpp::fcl::BVHModelBase>') to 'const boost::shared_ptr<hpp::fcl::CollisionGeometry>' for 1st argument
    shared_ptr & operator=( shared_ptr const & r ) BOOST_SP_NOEXCEPT
                 ^
[... cut ...]
/usr/include/boost/smart_ptr/shared_ptr.hpp:683:18: note: candidate function not viable: no known conversion from 'hpp::fcl::BVHModelPtr_t' (aka 'shared_ptr<hpp::fcl::BVHModelBase>') to 'boost::detail::sp_nullptr_t' (aka 'nullptr_t') for 1st argument
    shared_ptr & operator=( boost::detail::sp_nullptr_t ) BOOST_SP_NOEXCEPT
                 ^
2 errors generated.
src/CMakeFiles/pinocchio.dir/build.make:89: recipe for target 'src/CMakeFiles/pinocchio.dir/parsers/urdf/geometry.cpp.o' failed
```

The compilation issue happens in both: gcc and clang.

jmirabel, 2022-03-21:
Overlooking at it, it looks like a missing declaration of a forward declared class (i.e. a missing header inclusion).

jcarpent, 2022-03-21:
This is expected. The current devel branch of hpp-fcl use std::shared_ptr because it is now C++11, while Pinocchio is still C++98, so relying on boost::shared_ptr.
The current devel of FCL is compliant with Pinocchio-3x, which will be released soon.

closed 2022-03-21 by jcarpent

jmirabel, 2022-03-21:
Indeed I overlooked at it :laughing: