stack-of-tasks/pinocchio #1621: Position and orientation of end-effector

arjung128, 2022-03-21:
I'm working with the frank panda robot:

```
Nb joints = 11 (nq=16,nv=15)
  Joint 0 universe: parent=0
  Joint 1 root_joint: parent=0
  Joint 2 panda_joint1: parent=1
  Joint 3 panda_joint2: parent=2
  Joint 4 panda_joint3: parent=3
  Joint 5 panda_joint4: parent=4
  Joint 6 panda_joint5: parent=5
  Joint 7 panda_joint6: parent=6
  Joint 8 panda_joint7: parent=7
  Joint 9 panda_finger_joint1: parent=8
  Joint 10 panda_finger_joint2: parent=8
```

When I do:

```
pin.forwardKinematics(model, data, q)
pin.updateFramePlacements(model,data)
```

then `data.oMf.tolist()` contains 25 frames. `model.frames[x].parent` for x [15:20] returns 8 -- which of these 6 frames should I look at for the end-effector position? Thanks!

jcarpent, 2022-03-21:
It seems you've added a free-flyer joint to your robot. Is it on purpose?

jcarpent, 2022-03-21:
For control, you should always refer to the parent joint (frame.parent) quantity to get the closest supporting joint.
Is it answering your question?

arjung128, 2022-03-21:
But `model.frames[frame_id].parent` for `frame_id` in `[15:20]` is `8` -- which of these do I look at?

jcarpent, 2022-03-21:
model.joints[8] corresponds to the supporting joint.

arjung128, 2022-03-21:
That gives me:

```
JointModelRZ
  index: 8
  index q: 13
  index v: 12
  nq: 1
  nv: 1
```
  
What would the frame_id be? Sorry I'm not following.

jcarpent, 2022-03-21:
What is the name of your end-effector? This is on your side that you should know this information.

arjung128, 2022-03-21:
Got it, thanks!

jcarpent, 2022-03-21:
Perfect.

closed 2022-03-21 by jcarpent