stack-of-tasks/pinocchio #1623: Documentation fo building/installing on windows

kousheekc, 2022-03-25:
Hi,

I would like to use the Pinocchio library on a windows system since I would like to generate a custom dll using the library for further use in my project. The documentation provides clear instructions for installing the library on Linux and Unix systems but the installation procedure is not clear on windows.

I am assuming that I would have to build from source. I have downloaded the necessary requirements (Eigen and Boost) however I am not clear about how to configure the cmake so that the installation is successful. Some clear step-by-step instructions for downloading the library would be very helpful.

Thank you so much in advance.

jcarpent, 2022-03-25:
Dear @kousheekc,

I would recommend you to use conda if possible.
```
conda install pinocchio -c conda-forge
```

jcarpent, 2022-03-25:
It will install all the dependencies you need for that.

closed 2022-03-25 by jcarpent

kousheekc, 2022-03-25:
Hi @jcarpent, thank you for the quick response. I have tried the conda approach and this was the issue I faced.

The library installed successfully, however when I tried to compile the example c++ code using the command provided in the documentation:
`g++ -std=c++11 overview-simple.cpp -o overview-simple $(pkg-config --cflags --libs pinocchio)`

I received the error:
`g++: error: $(pkg-config: No such file or directory
g++: error: pinocchio): No such file or directory
g++: error: unrecognized command line option '--cflags'
g++: error: unrecognized command line option '--libs'`

Therefore I tried to manually include the directories using the -I approach using the following command:
`g++ overview-simple.cpp -o overview-simple -I C:\Users\kousheek\anaconda3\Library\include -I C:\Users\kousheek\anaconda3\Library\include\eigen3`

However I received the following error:
`c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/as.exe: C:\Users\kousheek\AppData\Local\Temp\ccHtYsBG.o: too many sections (34705)
C:\Users\kousheek\AppData\Local\Temp\ccXFygjw.s: Assembler messages:
C:\Users\kousheek\AppData\Local\Temp\ccXFygjw.s: Fatal error: can't write 397 bytes to section .text of C:\Users\kousheek\AppData\Local\Temp\ccHtYsBG.o: 'file too big'
c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/as.exe: C:\Users\kousheek\AppData\Local\Temp\ccHtYsBG.o: too many sections (34705)
C:\Users\kousheek\AppData\Local\Temp\ccXFygjw.s: Fatal error: can't close C:\Users\kousheek\AppData\Local\Temp\ccHtYsBG.o: file too big`

I have never encountered this error before and can't find any documentation online to resolve this.

What could I do to resolve this problem.
Thank you so much.

jcarpent, 2022-03-25:
You have to use the `/bigobj` compilation flag

kousheekc, 2022-03-25:
This worked thank you.

kousheekc, 2022-04-17:
Hi @jcarpent , unfortunately, I am facing another issue...

I have tried compiling the first example in the documentation, i.e:
`#include "pinocchio/parsers/sample-models.hpp"
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/rnea.hpp"
 
int main()
{
  [pinocchio::Model](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1ModelTpl.html) model;
  [pinocchio::buildModels::manipulator](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio_1_1buildModels.html#accd83f24a102dbe4845cedc1c9a52f4b)(model);
  [pinocchio::Data](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1DataTpl.html) data(model);
 
  Eigen::VectorXd q = [pinocchio::neutral](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#ac1ea82f003a9814a0b1e127d42d32d73)(model);
  Eigen::VectorXd v = Eigen::VectorXd::Zero(model.nv);
  Eigen::VectorXd a = Eigen::VectorXd::Zero(model.nv);
 
  const Eigen::VectorXd & tau = [pinocchio::rnea](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a8db0fecde513356bf61dfe2dee2e6c7f)(model,data,q,v,a);
  std::cout << "tau = " << tau.transpose() << std::endl;
}`
and it compiles fine.

However, when I try to compile the very next example, i.e:
`#include "[pinocchio](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html)/parsers/urdf.hpp"
 
#include "pinocchio/algorithm/joint-configuration.hpp"
#include "pinocchio/algorithm/kinematics.hpp"
 
#include <iostream>
 
// PINOCCHIO_MODEL_DIR is defined by the CMake but you can define your own directory here.
#ifndef PINOCCHIO_MODEL_DIR
  #define PINOCCHIO_MODEL_DIR "path_to_the_model_dir"
#endif
 
int main(int argc, char ** argv)
{
  using namespace pinocchio;
  
  // You should change here to set up your own URDF file or just pass it as an argument of this example.
  const std::string urdf_filename = (argc<=1) ? PINOCCHIO_MODEL_DIR + std::string("/example-robot-data/robots/ur_description/urdf/ur5_robot.urdf") : argv[1];
  
  // Load the urdf model
  [Model](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1ModelTpl.html) model;
  [pinocchio::urdf::buildModel](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio_1_1urdf.html#a9222e9df7754648584079ac37836c163)(urdf_filename,model);
  std::cout << "model name: " << model.name << std::endl;
  
  // Create data required by the algorithms
  [Data](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1DataTpl.html) data(model);
  
  // Sample a random configuration
  Eigen::VectorXd q = [randomConfiguration](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a8aa12c4444aa4425160cd60218a112d3)(model);
  std::cout << "q: " << q.transpose() << std::endl;
 
  // Perform the forward kinematics over the kinematic tree
  [forwardKinematics](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/namespacepinocchio.html#a97be7e9cd332a591b1431e30bab2c51e)(model,data,q);
 
  // Print out the placement of each joint of the kinematic tree
  for(JointIndex joint_id = 0; joint_id < (JointIndex)model.njoints; ++joint_id)
    std::cout << std::setw(24) << std::left
              << model.names[joint_id] << ": "
              << std::fixed << std::setprecision(2)
              << data.oMi[joint_id].translation().transpose()
              << std::endl;
}`

I receive the following error:
``

traversaro, 2022-04-19:
Hi @kousheekc, I am not sure, but I am afraid that you are trying to use with the `x86_64-w64-mingw32` compiler a library compiled with MSVC. C++ libraries that are compiled with MSVC can't be be linked with mingw due to the fact that the symbol name mangling rules are different, that I think is the reason why the symbol  `__imp__ZN9pinocchio4urdf7details13parseRootTreeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS1_18UrdfVisitorBaseTplIdLi0EEE` is not found in the library. See https://stackoverflow.com/questions/2529770/how-to-use-libraries-compiled-with-mingw-in-msvc for a related link.

kousheekc, 2022-04-20:
Hi @traversaro, I am trying to use visual studio this time to make use of the MSVC compiler, however, I am confused about how to set the include directories in the project properties. I began by including the following into the "additional include directories":

```C:\Users\kousheek\anaconda3\Library\include```

and the following to the "additional library directories"

```C:\Users\kousheek\anaconda3\Library\lib```

 
But I received the following error:

```Cannot open include file: 'Eigen/Core': No such file or directory```

Next I included the following directory to try and solve the error:

```C:\Users\kousheek\anaconda3\Library\include\eigen3```

but received a bunch of other compilation errors.

I believe I am messing up somewhere with the includes. Any guidance would be very helpful.

Thank you so much.

traversaro, 2022-04-20:
Hi @kousheekc, this is probably out of scope here but you should not build a Visual Studio Solution from scratch, but rather use CMake either directly from Visual Studio (https://docs.microsoft.com/en-us/cpp/build/cmake-projects-in-visual-studio?view=msvc-170) or by using CMake to generate a Visual Studio solution. In this way you can use directly pinocchio's via `find_package(pinocchio REQUIRED)`.

kousheekc, 2022-04-26:
Hi @traversaro, I am now trying to use CMake to build a project but I am confused about how "find_package" can be used. I have used conda to install pinocchio and my current CMakeLists.txt looks like this:

```
cmake_minimum_required (VERSION 3.8)

project ("test_pinocchio")

set (CMAKE_CXX_STANDARD 17)

find_package(pinocchio REQUIRED)

# Add source to this project's executable.
add_executable (test_pinocchio "pinocchio.cpp" "pinocchio.h")

# TODO: Add tests and install targets if needed.
target_link_libraries(test_pinocchio PUBLIC pinocchio::pinocchio)
```
However, under the find_package line I receive the message:

```
Package 'pinocchio' not found
```

traversaro, 2022-04-27:
Are you sure you are running CMake in a command line in which first you activated the environment in which pinocchio is installed?

kousheekc, 2022-04-27:
@traversaro, I am running CMake within the environment in which pinocchio is installed. The CMake generates a solution but when I try to open and run the solution, I get the following error:

```the code execution cannot proceed because boost_serialization.dll was not found. Rainstalling the program may fix this problem.```

I am new to using CMake and conda environments on windows, please excuse me. Thank you.

traversaro, 2022-04-27:
Are you opening the Visual Studio solution in a terminal where the environment is activated?

traversaro, 2022-05-02:
I think you need to compile in `Release` mode, `conda-forge` does not `Debug` binaries.

jcarpent, 2022-05-04:
Thanks @traversaro for your precious help.