stack-of-tasks/pinocchio #1626: Inverse Kinematics result does not satisfy joint limits

arjung128, 2022-03-29:
Hello,

I'm doing inverse kinematics using the exact code from the examples, and the solutions I'm getting don't seem to satisfy the joint limits specified in the URDF file. Here's a minimal example:

```
import pinocchio
import numpy as np
from numpy.linalg import norm, solve

# load model
model, geom_model, visual_model = pinocchio.buildModelsFromUrdf('../data/panda/panda.urdf', '/../data/panda/')
data, geom_data, visual_data = pinocchio.createDatas(model, geom_model, visual_model)

# starting configuration
q = np.array([-8.60531747e-01, 1.48915422e+00, 1.47712374e+00, -6.70803905e-01, 3.77983612e-04, 3.00506473e+00, 7.79607475e-01, 4.54281690e-03, 3.69776948e-03])

# get finger gripper pose
JOINT_ID = 9
pinocchio.forwardKinematics(model, data, q)
oMdes = data.oMi[JOINT_ID].copy()

# move finger gripper slightly along x-dir
oMdes.translation -= np.array([0.4, 0, 0])

# rest is copied exactly from https://github.com/stack-of-tasks/pinocchio/blob/master/examples/inverse-kinematics.py
q      = pinocchio.neutral(model)
eps    = 1e-4
IT_MAX = 1000
DT     = 1e-1
damp   = 1e-12

i=0
while True:
    pinocchio.forwardKinematics(model,data,q)
    dMi = oMdes.actInv(data.oMi[JOINT_ID])
    err = pinocchio.log(dMi).vector
    if norm(err) < eps:
        success = True
        break
    if i >= IT_MAX:
        success = False
        break
    J = pinocchio.computeJointJacobian(model,data,q,JOINT_ID)
    v = - J.T.dot(solve(J.dot(J.T) + damp * np.eye(6), err))
    q = pinocchio.integrate(model,q,v*DT)
    if not i % 10:
        print('%d: error = %s' % (i, err.T))
    i += 1

if success:
    print("Convergence achieved!")
else:
    print("\nWarning: the iterative algorithm has not reached convergence to the desired precision")

print('\nresult: %s' % q.flatten().tolist())
print('\nfinal error: %s' % err.T)
```

When I run this, the last element of the result, `q[-1]`,  is `-0.20580117405759688`, which is outside of the `[0, 0.04]` joint limits specified in the URDF file. Is there anything additional I need to do such that the solutions satisfy the joint limits? Or am I missing something?

Thanks in advance for the support!

jcarpent, 2022-03-30:
As already mentioned in #1624, you have to use a QP solver to enforce the bound constraints.

closed 2022-03-30 by jcarpent

arjung128, 2022-03-30:
Hi @jcarpent,

I'm sorry but I don't see where this was mentioned in #1624. The only instance of QP in #1624 seems to be from code that I posted, which I got from #1336. Is this what you were referring to?

I modified the above code to use a QP solver (based on the code from #1336). Here are the changes I made:

```
# J = pinocchio.computeJointJacobian(model,data,q,JOINT_ID)
mu = lm_damping * max(1e-3, err.dot(err))
pinocchio.computeJointJacobians(model,data,q)
J = pinocchio.getJointJacobian(model, data, JOINT_ID, pinocchio.LOCAL)
ATA = J.T @ J + mu * np.eye(model.nv)
ATb = -err.T.dot(J)

# v = - J.T.dot(solve(J.dot(J.T) + damp * np.eye(6), err))
v = -solve_qp(ATA, ATb)
```

While this converges, the last element of the result still seems to be out of bounds. Is this what you were suggesting? I'm sorry if I misunderstood your suggestion.

Thanks for all your help!

jmirabel, 2022-03-30:
Inverse kinematics involves numerical optimization. Without joint bound, it is a very simple problem. With joint bound, it is a little harder and it requires you (i.e. someone willing to write an IK solver) to have some numerical optimization understanding. 
 
Given your answers, I guess you are not familiar with numerical optimization. Before even trying to code anything, I suggest you try to understand theoretically what a QP is, what you want to do and how you can formulate what you want as a QP.

@jcarpent already gave you a hint to get you started. There are plenty open-source implementations of IK solver with bound constraint if you need inspiration (e.g. one not using QP: github.com/humanoid-path-planner/hpp-constraints).

Mind that numerical optimization is a bit out of scope in this repo.

olivier-stasse, 2022-03-30:
We have already a repo implementing Inverse Dynamics and they are some ongoing effort to add Inverse Kinematics too.
The repo is available here:
https://github.com/stack-of-tasks/tsid

If you want to do model predictive control:
https://github.com/loco-3d/crocoddyl

jcarpent, 2022-03-30:
> I'm sorry but I don't see where this was mentioned in #1624. The only instance of QP in #1624 seems to be from code that I posted, which I got from #1336. Is this what you were referring to?

I was referring to #1619

arjung128, 2022-04-11:
@jmirabel Thanks so much for the detailed response! Enforcing the joint limits in the numerical optimization loop seemed to resolve the issue, too.

@olivier-stasse Thanks for the pointers!

@jcarpent Got it, thanks!