stack-of-tasks/pinocchio #1632: Collision checking with point cloud in world coordinate frame

arjung128, 2022-04-11:
Hello,

I am trying to detect collisions between a point cloud (in the world coordinate frame) and the robot. It seems like the collision checking example ([here](https://github.com/stack-of-tasks/pinocchio/blob/master/examples/collisions.py)) is checking for self-collisions -- how can this example be extended to check for collisions with an arbitrary point cloud in the world coordinate frame?

Apologies if this is a naive question, I am not very familiar with Pinocchio / FCL. Any help will be much appreciated, thanks in advance!

jcarpent, 2022-04-13:
Could you provide an example of a point-cloud?
I will add a short tiny example using Pinocchio and hpp-fcl.

arjung128, 2022-04-13:
I have object meshes (with file type `.obj` / `.glb`) which I would like to place at various positions / orientations and detect collisions with the Franka panda robot. I was thinking of converting these meshes into 3D point clouds (i.e. a numpy array of 3D points in the world coordinate frame) to be used for collision checking... unless there's another representation which is better suited for this task?

Thanks for your help!

jcarpent, 2022-04-13:
OK. Could you give me an example of such a point cloud?
Best,

Justin

arjung128, 2022-04-13:
For example, I created a point cloud of a box with dimensions 1x1x1 using: `verts = np.random.rand(3, 100000)`. This box currently has one corner at the origin, but if I were to move this box around to an arbitrary position / orientation, I'd still like to detect collisions against an arbitrary pose of the robot with it. 

I hope this is clearer!

closed 2022-04-25 by jcarpent